SynthstromAudible/DelugeFirmware #472: Loop on waveform view not working

moulfrit, 2023-08-29:
### Please describe the problem:

In waveform view loop shortcut ( green / red bar + any pad ) doesn’t not working .

### What is the expected behavior?

Make loop section

### Is there a relevant Pull request?

I don’t know 

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

nightly

### What is the firmware name:

Deluge-v1_0_0-nightly+2023_08_29-dbaa842.bin

### If possible provide the steps to reproduce the issue and upload additional media:

1.open waveform view

2.press green bar+ any pad

result: nothing is happens

m-m-adams, 2023-08-29:
I'm troubleshooting a bit and if you zoom in far enough that the loop point and the start point no longer overlap you can move them around properly. Looks like the bug might be that there's a loop point by default for some reason

stellar-aria, 2023-10-13:
Closing as fixed by #619

PR #619 (2023-10-13): Bugfix/toggle between sample markers on column
Fix #472 

If sample loop points are very close to the start point then neither can be edited without zooming in. This PR changes the behaviour so that repeated presses toggle between the markers sharing that column.
PR #618 (2023-10-13): Bugfix/remove sneaky optimization and replace with virtual functions
The sneaky optimization seems to have resulted in issues around multi samples (which I think is actually all samples internally)

With this fix I can no longer reproduce the following bugs:

Fix #610 
Fix #492
~~Fix #472~~ still happens sometimes
closed 2023-10-13 by stellar-aria