SynthstromAudible/DelugeFirmware #481: Grid View, display hangs a bit before showing Clip information.

ok-reza, 2023-08-30:
### Please describe the problem:

When in Grid View, when you press and hold a clip, the title and type of clip is shown (via display and lit clip; synth, kit, midi, etc).

However, when compared to normal Song View (rows), this isn't as instantaneous, there is a little lag. This makes navigating clips feel more sluggish.

### What is the expected behavior?

Clip info should appear as fast as Song View (rows).

### Is there a relevant Pull request?

#251

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

nightly

### What is the firmware name:

ac1002e

PaulFreund, 2023-08-30:
This is an intentional wait time to distinguish arming/disarming and "selecting" clips. I will make a test version with the rightmost sidebar pads as function selectors but I expect the learning curve to be higher.

PaulFreund, 2023-09-06:
Please check #500 which should fix your issue since in Edit mode the selection comes immediately

ok-reza, 2023-09-08:
i'll close this since things have shifted in implementation.

i will say that the sluggishness wouldn't have been an issue if the display info was immediate, even if behind the scenes there is a delineation between arming vs selecting clips.

closed 2023-09-08 by ok-reza