SynthstromAudible/DelugeFirmware #482: Grid View, entering clip flashes "Song" pad before switching back to "Clip" pad

ok-reza, 2023-08-30:
### Please describe the problem:

When you hold down a clip in Grid View, the 'clip' pad is illuminated. Once you hit 'clip' to enter it, during the animation of entering the clip, 'song' pad is illuminated briefly before permanently switching to 'clip' being illuminated.

### What is the expected behavior?

Like the behavior in Row Song View, the 'clip' pad should remain illuminated during animation into entering the clip.

### Is there a relevant Pull request?

#251

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

nightly

### What is the firmware name:

2023_08_28-ac1002e

### If possible provide the steps to reproduce the issue and upload additional media:

Described above

PaulFreund, 2023-09-06:
Please check #500, if I understood you correctly this shouldn't be an issue anymore since clip button is no longer used

stellar-aria, 2023-09-17:
Closing this since no longer able to replicate. (Fixed)

closed 2023-09-17 by stellar-aria