SynthstromAudible/DelugeFirmware #488: kind of bug ?

melokit, 2023-08-31:
### Please describe the problem:

So,when I want to load a song (load+select+scroll) I have to go from the end to the beginning (backward) because in the other direction (forward) the selection loops over 3/4 songs and cannot to go beyond..
there is also, as for other people apparently, the RESAMPLE function which stops working in full recording (to make a STEM for example). I have to restart the deluge, I can do 2/3 stems again and it starts again

### What is the expected behavior?

-to be able to navigate my songs in both directions
-that the resample function is stable and does not stop in the middle of a recording

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

nightly

### What is the firmware name:

1.0.0-nightly-b4aeb97

### If possible provide the steps to reproduce the issue and upload additional media:

next time

m-m-adams, 2023-09-01:
I can't reproduce the first one, maybe your encoder is faulty?

I also can't reproduce the second one, could you add some more detail on what you're resampling? It might be related to the song contents, especially if they're using large samples there are known bugs that were working on fixing

m-m-adams, 2023-09-01:
I was able to reproduce the second bug using more audio samples

m-m-adams, 2023-09-25:
This should be fixed now (or at least less common)

closed 2023-10-07 by m-m-adams