SynthstromAudible/DelugeFirmware #490: USB Midi clock running too fast and inconsistent (double clock)

romor001, 2023-08-31:
### Please describe the problem:

Attached Deluge to Macbook via USB that is running Protokol (https://hexler.net/protokol). 

Deluge appears to be sending Midi clock from two of the three USB Midi devices it provides, resulting in a double clock that is instable and fluctuating.

[image]

Cross check with Release 4.1.3 firmware shows correct behaviour:
[image]

In reverse when receiving Midi clock something similar happans, clock is roughly doubled and fluctuating.

Discussed with @ilovesteak in #nightly-testing on Discord.
Bug seems to be a regression from splitting the USB-Midi device into three devices. The behaviour can be desirable when using UDB-Midi functionality with multiple receiving devices (MPE and non-MPE).
Proposed solution is a configurable parameter in the Midi Clock Out section with parameters port 1/2/both/off.

### What is the expected behavior?

The expected behaviour is to send Midi clock only once.
As described, sending clock on both USB-Midi devices can be desirable, thus a user selectable configuration has been proposed.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

nightly

### What is the firmware name:

deluge-v1_0_0-nightly+2023_08_29-dbaa842.bin

### If possible provide the steps to reproduce the issue and upload additional media:

See description.

m-m-adams, 2023-08-31:
Thanks for the reminder! It's @m-m-adams on GitHub although I appreciate that someone else is living the Ilovesteak life

stellar-aria, 2023-10-13:
Closing as fixed by #620

PR #620 (2023-10-13): Bugfix/duplicate midi clocks
Fix #490 

Creates a per midi device setting to choose whether to send clocks on that device
closed 2023-10-13 by stellar-aria