SynthstromAudible/DelugeFirmware #491: Performance issues with noticable sample cutout

PaulFreund, 2023-08-31:
### Please describe the problem:

Performance seems to be degraded which is especially noticable for sample heavy tracks since sample cutout is very noticable.

Reproduction is  done with this song: [CORDAE](https://discord.com/channels/608916579421257728/991432258286268496/1146194461626474600)

### What is the expected behavior?

The song can be played in full without cutouts with current Official Beta and e2 generated v4.1.4-alpha3 commit. 

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

nightly, community, Built from community, Built from release/1.0, other

### If possible provide the steps to reproduce the issue and upload additional media:

1. Copy https://discord.com/channels/608916579421257728/991432258286268496 to your Songs folder
2. Load either firmware with Load/Run OLED Release
3. Load song (CORDAE) - Arranger is opening
4. Play the song and listen, tracks are indexed from 1 top to bottom
    * Latest Rohan beta/release or 4.1.4-alpha3 with e2: All tracks play to the end without manual interaction with the device
    * fa9ab593: It is normal that track 3 (rap acapella) is cutting out 8/10 times all other sounds should continue to play through
    * 5f80c5c4: 
        * track 3 (acapella) is cutting out between pad 5 to 7 from the left
        * track 5 (bass) is cutting out at some point
        * track 6 (bells) is cutting out shortly after starting every time
        * track 7 (pad) can also cutout

PaulFreund, 2023-08-31:
There seem to be three known sources of decreased performance currently:
1. #350 seems to be happening once internal memory allocation for objects in the audio path fails and they are moved to SDRAM
2. From Scons (fa9ab59) to CMake (5f80c5c) there seems to be a quite big delta
3. From e2 (GCC9) to dbt (GCC10/12) there seems to have been a smaller but noticable delta too 

Better performance tracing will be of great use to get back or exceed preexisting performance

Icoustic, 2023-08-31:
Seeing the same when comparing with official beta on elaborate songs with lots of audio clips/ samples

m-m-adams, 2023-09-10:
Should we close this out or find a new reference song? Cordae no longer cuts out in nightly builds

stirman, 2023-09-12:
Confirmed. I can't reproduce this issue with the latest nightly build as of today. Thx devs!!! <3

Icoustic, 2023-09-13:
> Should we close this out or find a new reference song? Cordae no longer cuts out in nightly builds

I can reproduce with this song of mine; plays normal on official beta, while in nightly build (11/9) audio clip loop cuts out.
Specifically, the bottom audio clip bassline (playing in the provided project), which is 64 bars long. On nightly build, it always cuts out at bar 62ish.
Starts playing again on the next 1, but is now "stunted" and can't even play as long as before, and keeps cutting out.

Link coming up

PaulFreund, 2023-09-22:
Fixed with #541

Icoustic, 2023-10-13:
> I can reproduce with this song of mine; plays normal on official beta, while in nightly build (11/9) audio clip loop cuts out. Specifically, the bottom audio clip bassline (playing in the provided [...]

This issue is back again now, current nightly.

PaulFreund, 2023-10-14:
Please test again with the new nightly, the last one had audio logging on by accident. Optimally test with current beta now since we work on getting that as stable as possible.

stirman, 2023-10-15:
I haven't been able to duplicate this issue with any nightly builds in the
last month or so.

—
Stirman
https://instagram.com/stirman

Icoustic, 2023-10-15:
> Please test again with the new nightly, the last one had audio logging on by accident. Optimally test with current beta now since we work on getting that as stable as possible.

Tested beta, yeah, back to the way it was now :)

stirman, 2023-10-26:
Sorry, was traveling. Back now, tested on today's nightly build, can't reproduce the issue with long samples. Can't reproduce the issue with the original "Cordae" track either.

Icoustic, 2023-10-28:
> This issue is back again now, current nightly.

I still see this issue with long audio clip dropping out and song subsequently getting messed up all around and dropping more voices @PaulFreund @m-m-adams 
Beta from 27th.

Icoustic, 2023-10-28:
And also the SD card error from #462 still happening, unlike official rel/ beta where it never happens as such

stirman, 2023-11-03:
Seeing this issue again, and it's cutting off much shorter samples now. Let me know if there's something specific I can help test. On nightly build from about a week ago.

PaulFreund, 2023-11-03:
:) Hey @stirman my general recommendation is to always test with the newest release before reporting as sometimes there can be intermittent issue. if you test now you will get a build without known issues like that. The only situation where I could get your song to drop out was when I completely filled the audio cache with multisample mashing before and then only one playthrough, the ones after that worked fine.

stirman, 2023-11-03:
Totally, I know better. I got today's build (2023_11_03-337c437) and I can't reproduce the issue again. Will continue trying with more songs/samples and report back if I can repro this issue. Thx again for all the work you community devs do, it's amazing!

Icoustic, 2023-11-09:
> > > > Should we close this out or find a new reference song? Cordae no longer cuts out in nightly builds
> > > 
> > > 
> > > I can reproduce with this song of mine; plays normal on official beta, while in nightly build (11/9) audio clip loop cuts out. Specifically, the bottom audio clip bassline (playing in the provided project), which is 64 bars long. On nightly build, it always cuts out at bar 62ish. Starts playing again on the next 1, but is now "stunted" and can't even play as long as before, and keeps cutting out.
> > 
> > 
> > This issue is back again now, current nightly.
> 
> I still see this issue with long audio clip dropping out and song subsequently getting messed up all around and dropping more voices @PaulFreund @m-m-adams Beta from 27th. Got it on video if needed

Just letting you know I can still reproduce the issue I mentioned in quote above, on latest beta downloaded last night.
Can't play 64 bar audio clip loop all the way thru, breaks other voices after.

m-m-adams, 2023-11-09:
Can you share the song? I want to look at exactly why it's dropping the voice

Icoustic, 2023-11-09:
> Can you share the song? I want to look at exactly why it's dropping the voice

Sure, will send in a few hours

Icoustic, 2023-11-10:
> Can you share the song? I want to look at exactly why it's dropping the voice

Here’s the song again, the long audio clip is the one with pink section color. Drops out after some time and then other clips are also affected. As mentioned before this all plays fine on the latest official Synthstrom release and 4.1.4 beta.

https://we.tl/t-s5xN1SBclm

m-m-adams, 2023-11-10:
I don't know if I'm playing it wrong but I'm playing the pink section and it's played through 4 times without dropping any voices so far. Are you sure you're on the latest beta and the update took effect?

m-m-adams, 2023-11-10:
Even after loading and playing other songs this song continues to play fine for me. Please let me know if there's something else that I need to do beyond playing the pink section

Icoustic, 2023-11-10:
> Even after loading and playing other songs this song continues to play fine for me. Please let me know if there's something else that I need to do beyond playing the pink section

Really? Is the pink one the one with 64 bars or did I mix it up? Did you play it together with other tracks?
I can reproduce it easily. Could it have somehting to do with using a larger SD card? I use 400GB. But as mentioned it plays fine on official firmware.

Icoustic, 2023-11-10:
> I don't know if I'm playing it wrong but I'm playing the pink section and it's played through 4 times without dropping any voices so far. Are you sure you're on the latest beta and the update took effect?
> 
> Cool song!

I was on the beta before yesterday's latest when I wrote it - wouldn't think it'd be fixed inbetween, but I need to double check in studio later if the latest is different.

Icoustic, 2023-11-11:
> Even after loading and playing other songs this song continues to play fine for me. Please let me know if there's something else that I need to do beyond playing the pink section

Well, good news, it is indeed working *most* of the time, but by loading a couple more sample heavy and close-to-the-ceiling songs (3-4 ish), it does happen. Tried reverting to beta from nov7, reproduced there couple of times with loading more stuff as mentioned. On beta from nov10, same thing, maybe a bit better, kinda hard to tell as the bug is a bit obscure, sneaky and sporadic. The type I hate 😂😭

I’m having a hard time reproducing those old SD card errors though at least, so that’s a good thing 🙂

m-m-adams, 2023-11-11:
Definitely this song would have had issues before the 2nd but I don't think anything would have changed since then. It's possibly very close to failing to play properly so that minor random differences in the memory layout lead to issues.

We have a few more optimizations for memory use in mind so hopefully they can get it working repeatedly!

Icoustic, 2023-11-11:
> Definitely this song would have had issues before the 2nd but I don't think anything would have changed since then. It's possibly very close to failing to play properly so that minor random [...]

Cool 👍 If needed (if you weren't able to reproduce), I can send a couple heavy songs so you can check it

m-m-adams, 2023-11-12:
Can you try this PR? https://github.com/SynthstromAudible/DelugeFirmware/pull/716 

The current build is here - https://github.com/SynthstromAudible/DelugeFirmware/actions/runs/6843170006

Icoustic, 2023-11-13:
> Can you try this PR? #716
> 
> The current build is here - https://github.com/SynthstromAudible/DelugeFirmware/actions/runs/6843170006

Sure, ASAP!

Icoustic, 2023-11-13:
> Can you try this PR? #716
> 
> The current build is here - https://github.com/SynthstromAudible/DelugeFirmware/actions/runs/6843170006

Issue still there 😕 Load and play some big songs, come back to the one I sent here, and most voices drop quickly from the start while the 64 bar audio clip isn’t even playing yet.
It just seems to stack up stuff in memory from past songs when loading new ones and doesn’t clear it out fast enough. Not that I know, I’m no coder. After restarting playback several times things seem to start playing right again, but that’s not exactly ideal behaviour

m-m-adams, 2023-11-13:
Would you mind checking the cluster size on your SD card?

Icoustic, 2023-11-14:
> Would you mind checking the cluster size on your SD card?

I'm not by my Deluge(s) now I'm pretty sure its 32kb, on my 400gb card. Will confirm later

PaulFreund, 2023-11-14:
@Icoustic If you want to test immediately you can try this build https://github.com/SynthstromAudible/DelugeFirmware/suites/18180188491/artifacts/1049081351 otherwise I can notify you once the PR is merged :)

Icoustic, 2023-11-14:
> @Icoustic If you want to test immediately you can try this build https://github.com/SynthstromAudible/DelugeFirmware/suites/18180188491/artifacts/1049081351 otherwise I can notify you once the PR is merged :) Should hopefully fix all the remaining dropouts

Checking it out now 👍

PaulFreund, 2023-11-16:
@Icoustic would be great to get an update :) Eagerly hoping for confirmation

Icoustic, 2023-11-16:
> @Icoustic would be great to get an update :) Eagerly hoping for confirmation

Hey, sorry for the wait & suspense! Been super busy and haven't had any time, but I'm going to test properly tonight and I'll report back

Icoustic, 2023-11-17:
@PaulFreund It seems to be good now! 👍👍 No irregular voice drops, and no “SD card error”s! Big ups 🥇 and thanks

PaulFreund, 2023-11-17:
Thank you so much for the feedback :) closing again for now!

PR #722 (2023-11-15): Bypass culling in case of cluster reclaiming and disabled sequentiality tests in alpha or beta verions
Should fix #491
closed 2023-11-17 by PaulFreund