SynthstromAudible/DelugeFirmware #492: Sample Preview shows incorrect waveform, file name, and audio playback isn't full length

entzmingerc, 2023-09-01:
### Please describe the problem:

With PLAY turned OFF (no green light)
With PREView set to ON, when browsing samples (audio files, .wav):
- On each new sample preview when turning select encoder to the right, the new sample audio might play for only half a second, the waveform displays either no waveform or the last correctly rendered waveform, not the correct waveform for the sample, and the file name on 7SEG displays the last correctly previewed sample and doesn't change when scrolling
- By switching back and forth between two samples in a folder, eventually the sample will playback for the full length correctly, eventually the correct waveform for the audio is rendered
- Eventually all samples in a folder can playback correctly with enough preview attempts
- Samples that might be previewing correctly will go back to being incorrect after switching to other folders and previewing other samples (unsure, hard to verify)

With PLAY turned ON (green light!)
- This behavior seems much less frequent if there at all
- A folder of incorrectly previewing samples seem to start previewing correctly
- Related to Issue #350, quickly scrolling samples while previewing can cull voices a lot quicker than they would usually be culled. 

It seems inconsistent but reproducible, tested on Cordae song and another song, two different nightly firmwares, multiple different folders of samples of varying lengths. 

### What is the expected behavior?

If PREView is set to ON in the System Settings, when turning the select knob to select an audio file, or when entering into a folder of audio files (.wav):
- it plays the audio file from start to finish
- the waveform for the current sample that is playing is displayed on the main grid pads
- its file name is displayed on the display

### Is there a relevant Pull request?

unknown

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

nightly

### What is the firmware name:

deluge-v1_0_0-nightly+2023_08_31-b4aeb97

### If possible provide the steps to reproduce the issue and upload additional media:

To reproduce, browse into a folder of samples, turn select encoder to the right and for each click of the encoder, listen for the preview playback to be cut out after a short amount of time, notice if the waveform displayed is the same as the previous sample previewed, and notice if the file name displayed is the same as the previous sample previewed.

If an incorrect previewed sample is found, turn select left and right one time each to confirm if that sample isn't loading correctly. Attempt previewing the sample again and again until eventually the sample starts to preview correctly. It takes usually 2-5 attempts to get a sample to preview correctly. 

Recorded a video here, hopefully the link works, file was too big to upload: https://photos.app.goo.gl/Ehmias3AUy92z3U57

m-m-adams, 2023-09-02:
Reproduced on 7seg at current head

entzmingerc, 2023-09-06:
When I went to browse for a sample it didn't refresh the screen and it showed the last displayed note sequence, then when I went to turn the select knob to the right, the animation played and then displayed the note sequence again instead of the expected waveform of the sample. 

just thought I'd make a comment to log it

m-m-adams, 2023-09-08:
Possibly related, after browsing through a large number of synth presets I start getting incorrect loads

entzmingerc, 2023-09-15:
on a selected sample that is showing previewing symptoms listed above, I get "RAM" displaying across the screen when I try to press select then choose between WAVE and MULTI and I select WAVE to load it as a wavetable. 

however I can't reproduce this consistently, but I can reproduce it. It seems to load/not load as a wavetable regardless if the sample has previewed successfully or not. It seems to load/not load as a wavetable regardless of if the waveform rendered correctly.

entzmingerc, 2023-09-16:
[image]

caught this today too while fishing, I could recreate it once again but it took a lot of scrolling, just updating the log :)

m-m-adams, 2023-09-29:
Is this still happening for you? I haven't noticed it in the last few days

m-m-adams, 2023-10-13:
Tentatively fixed - please re open if you still experience this

entzmingerc, 2023-10-18:
Didn't do extensive testing, but definitely got several cases of the failed waveform rendering while browsing samples on 7seg last night while using the most recent beta. Similar patterns of solving it seemed to work like just scrolling back and forth until it eventually returned the correct waveform, or it'd just load the previously rendered grid LEDs. Like if it didn't work immediately it'd do the animation then display the clip view notes as "the waveform" across the grid. If it did successfully render a waveform, and then if I scrolled to the right then failed to render the new waveform, it'd just redisplay the last successfully rendered waveform. Scrolling back and forth between samples eventually got it all rendered correctly. Not sure if the audio was getting culled but these were just just drum sample one-shots so I didn't notice. 

rambling for ya, super appreciative of the work you've been putting into stabilizing the system, it's so cool

m-m-adams, 2023-10-25:
This seems to work consistently for me - I can't trigger the bug anymore

entzmingerc, 2023-10-30:
downloaded latest beta
gave it a whirl after loading like 10 kits on two different songs
not coming up, looks like the memory fixes fixed it :)

PR #618 (2023-10-13): Bugfix/remove sneaky optimization and replace with virtual functions
The sneaky optimization seems to have resulted in issues around multi samples (which I think is actually all samples internally)

With this fix I can no longer reproduce the following bugs:

Fix #610 
Fix #492
~~Fix #472~~ still happens sometimes
closed 2023-10-30 by entzmingerc