SynthstromAudible/DelugeFirmware #501: Drawn MIDI curves are steppy on hardware syntzs

MegaloMIDIac, 2023-09-02:
### Please describe the problem:

If you draw a MIDI curve with the pads on a MIDI CC, it will sound steppy. If you record MIDI CC from the external synth all is fine. 

### What is the expected behavior?

Smooth transitions 

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

nightly

m-m-adams, 2023-09-02:
Can you enable interpolation on the external synth? 

Interpolating all midi to a high resolution on the deluge would significantly increase the amount of CC data sent which 
1) impacts note on/off timing since it's a serial protocol
2) requires more cpu time in the deluge

The recording works fine because your synth is sending CCs at a higher rate than the deluge automation currently generates them (1 per step at current zoom level). We can play with that frequency a little but without interpolating externally it will always be a bit steppy

MegaloMIDIac, 2023-09-02:
I have tested this behaviour with the Microfreak, Ambika and Blofeld. I don't think they can interpolate incoming MIDI CC.
You can hear this especially if you use the filter cutoff CC.

seangoodvibes, 2023-09-02:
This is with interpolation on in automation view right?

How are you inputting the MIDI automation lanes on the deluge? 

If you are changing one step at a time, it will be stepped regardless of whether interpolation is on or off. 

But if you enter a long press (press two pads to connect them) it will be interpolated when interpolation is on and stepped when interpolation is off.

MegaloMIDIac, 2023-09-02:
I tried both, with and without interpolarion.
Also the long press will get you the steppy result unfortunately.

seangoodvibes, 2023-09-02:
Thanks for reporting this @MegaloMIDIac 

I just confirmed the same behaviour sending MIDI to my dirtywave m8.

Hopefully we can find a solution for this as part of our first release as this is a notable bug.

Cheers

Sean

m-m-adams, 2023-09-02:
Yeah the automation view sets a value for each pad at the current zoom level. Properly smoothing to only change cc value by 1 at a time will require deeper changes and still sound steppy on many synths