SynthstromAudible/DelugeFirmware #503: Song previews display inverted when loading a new song.

0beron, 2023-09-03:
### Please describe the problem:

Once a song is loaded, pressing [LOAD] once more goes back to the song browser, however the preview image on the pads for the current song is displayed upside down. Navigating away and back to the same song then displays it in the original orientation.

### What is the expected behavior?

Song preview orientation should be consistently oriented whenever shown in the song browser.

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

Built from release/1.0

### What is the firmware name:

4c0bf781877e905d8b96e628d75baf4d30a33682

### If possible provide the steps to reproduce the issue and upload additional media:

1. Load any song (well, one with a preview image where you can tell if it's inverted or not!). Pads match the preview image since it is freshly loaded.
2. Press [LOAD] to return to the song browser. Pads dim and show the preview image again, but observe it is inverted.
3. Turn [SEL] to navigate one song left or right, and back again, image is now right-side-up.

ok-reza, 2023-09-05:
confirmed on my end too, the animation is also inverted vertically.

stellar-aria, 2023-09-17:
Fixed with #551

PR #551 (2023-09-17): Fix vertical scroll inversion issue
Fixes #503.

[Originally](https://github.com/SynthstromAudible/DelugeFirmware/blame/596e38a64842f21da5cbcb23718431b4500dad0d/src/deluge/hid/led/pad_leds.cpp#L1297) the comparison was greater than _anyways_, but somehow got swapped in the PIC refactor.
closed 2023-09-17 by stellar-aria