SynthstromAudible/DelugeFirmware #510: Slicer mode doesn’t result in clean transitions between chops

trappar, 2023-09-04:
### Please describe the problem:

When using slicing, the default settings do not allow for clean sounding manipulation of slices. See reproduction for video example.

### What is the expected behavior?

The default settings for slicing should allow for the sample to be played back by sequencing the slices together in order without any audible difference from simply playing the sample.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

nightly, other

### What is the firmware name:

4.1.3

### If possible provide the steps to reproduce the issue and upload additional media:

[Reproduction video](https://youtu.be/1_G7XmN3VEA)

trappar, 2023-09-04:
> I think this is because the slicer sets the trigger mode to `CUT` by default

Nope. It sets to once by default.

trappar, 2023-09-04:
After some discussion in the discord, here's my suggestion for how to handle this:

1. Sample playback ONCE mode needs zero crossing detection (which it for some reason doesn't have  despite all other modes having this)
2. When a new kit is created by slicing:
  * Set attack and release to 0
  * Set polyphony to choke

Nice thing about this is that it would only affect defaults of newly created sliced kits, so there's no need for a community features option.

I'm still not sure that it makes sense for sample playback mode to depend on slice length specifically in the context of slicing, but I think it's fine. If it gets set to ONCE you'd have a highly playable / sequenceable kit. If it gets set to CUT it would still work great when sequenced, it would just be harder to play live.

m-m-adams, 2023-10-14:
Closing but open to PRs addressing it. This is really a feature request

closed 2023-10-14 by m-m-adams