SynthstromAudible/DelugeFirmware #523: Switching from Drum Keyboard View to Synth Keyboard View will always make Synth Keyboard View isometric.

ok-reza, 2023-09-06:
### Please describe the problem:

If you set the synth keyboard layout for a particular clip to 'In-Key' and are currently in that view then switch to kit clip (press KIT, drum keyboard view is displayed), then switch _back_ to synth clip, the keyboard view is reset to 'Isomorphic', not respecting what the clip was originally set to.

### What is the expected behavior?

Switching between clip types while in keyboard view should maintain the Keyboard View layout. For instance, if you are in Drum Keyboard View and you zoom to have 2x2 squares, if you press SYNTH and then back to KIT, the same zoom level of Drum Keyboard View is maintained. Similarly, keyboard layout should do the same.

### Is there a relevant Pull request?

#517

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Built from release/1.0

### What is the firmware name:

2023_09_06-57c4c78.bin

### If possible provide the steps to reproduce the issue and upload additional media:

1) New Project
2) Enter Keyboard View of Synth Clip
3) Hold KEYBOARD pad + scroll with select encoder to switch from 'Isomorphic' to 'In-Key'
4) Hit 'KIT' to switch to kit clip (drum keyboard view will appear)
5) Hit 'SYNTH' to switch back to synth keyboard view: 'Isomorphic' keyboard view is now selected instead of the last selected view; 'In-Key'

PaulFreund, 2023-09-11:
I had the expectation that changing clip type might lose some information so the current implementation is not able to remember the last keyboard type that was compatible. Can we mark this as undesireable and slate an improvement for next release?

m-m-adams, 2024-03-03:
We're going to just block changing to/from kit if there's notes, which should make this irrelevant

seangoodvibes, 2024-03-05:
> We're going to just block changing to/from kit if there's notes, which should make this irrelevant

Might still be relevant because without notes you would still be able to switch from synth to kit and vice versa and the issue mentioned would still exist about it not remembering what your selected layout was. 

Not sure why you'd switch instrument types like that tho. Maybe by mistake?

sichtbeton, 2024-03-22:
Not true anymore. It remembers this. Could be closed?

seangoodvibes, 2024-03-22:
> Not true anymore. It remembers this. Could be closed?

It still happens. Try changing synth keyboard to in key and then switch to kit and back. it will go back to isomorphic layout.

m-m-adams, 2024-04-04:
We've now blocked switching if notes are entered, which makes this a moot point imo

closed 2024-04-04 by m-m-adams