SynthstromAudible/DelugeFirmware #524: Switching a clip to Kit Keyboard View then going to Song Row view will make the Kit clip not maintain Keyboard View..

ok-reza, 2023-09-06:
### Please describe the problem:

If you switch a clip from being a Synth/Midi/CV clip (with keyboard view on) to a Kit Clip _then_ exit the clip by hitting 'Song', the kit clip has now reverted back to sequencer view instead of maintaining it's keyboard view. If you enter that kit clip, you will see that it is not in keyboard view anymore.

### What is the expected behavior?

When you exit an empty clip in keyboard view (to song Row view), you will see the clip row colored as a rainbow to indicate it's currently in keyboard view. When you enter the clip again, you will be brought to the keyboard view of that clip.

When notes are sequenced in a clip already, you won't see the rainbow row in Song view, but the same behavior will be maintained: the last view you were in for that clip will maintain when you enter back into it.

### Is there a relevant Pull request?

#517 but most likely something older

### What hardware did you reproduce it with?

OLED, 7SEG

### What firmware did you reproduce it with?

nightly

### What is the firmware name:

2023_09_05-593bfc5 & 2023_09_06-57c4c78

### If possible provide the steps to reproduce the issue and upload additional media:

1) Enter blank Synth Kit
2) Enter Keyboard View
3) Switch clip type to KIT
4) Exit to SONG view (song row view)
5) Note that there is no rainbow colored clip to indicate Keyboard view being active for that empty clip.
6) Re-enter Kit clip
7) Note that you are no longer in Drum Keyboard View, but the sequencer view now.

Alternatively, you can start in a blank Kit Clip, switch type to Synth clip, then switch type  _back_ to Kit Clip and the same issue will occur. **Arriving at kit keyboard view at least once is what triggers this.** Meaning, you can switch from SYNTH, KIT, CV (all with keyboard view active), and since you've visited KIT clip once, when you exit to SONG view the clip will lose it's active Keyboard View when reentering.

bobtwinkles, 2023-09-07:
The architectural fix for this is probably to break `InstrumentClip` in to a component for the instrument itself and a component for the UI state

ok-reza, 2023-09-07:
From what I can tell, this is normally the case. Usually the deluge remembers the clip's view and zoom state when entering in and out of different clips (as far as I can tell). My issue seems to be a way to break this behavior from the norm.

seangoodvibes, 2023-09-07:
This should be an easy fix. To know whether to go back to Keyboard View it relies on a "onKeyboardScreen" variable which is a component of the InstrumentClip class.

@PaulFreund should be able to fix this relatively easily :)!

bobtwinkles, 2023-09-18:
Should be fixed in the next nightly.

PR #537 (2023-09-18): Fixed keyboard being closed after switching to kit mode
Fixes #524
closed 2023-09-18 by bobtwinkles