SynthstromAudible/DelugeFirmware #525: Highlighting incoming Midi notes can be improved for further clarity

ok-reza, 2023-09-07:
### Please describe the problem:

This feature is very nice for incoming midi. I think there are some clarity issues that could be improved on. My perspective is without having a norns so I may be unaware of some important use-cases.

I noticed that the pads light up based on the velocity of the incoming notes. 
This means that lightest touches will not light up pads at all. If you press a midi key with medium pressure, it will match the non-octave notes making it hard to tell what was highlighted (scale mode on). Additionally, since root octave notes are fully lit in keyboard view already, you won't get visual feedback when hitting root notes. 

### What is the expected behavior?

Possible solutions/improvements:

- Make the highlighted notes always white (or a designated color the user choses in Pads>Colors menu).
- Make the lowest velocity highlight possible still turn on the LED for that pad.
- Or simply flash the start of an incoming note white when first pressed then resume current pad highlight behavior.
- Have no velocity pad brightness influence at all. (Toggleable option in menu?)

### Side Note
I think this behavior is great and eventually should be implemented during regular playback when you are viewing a clip that already has sequenced notes. This would allow users to add notes to an already playing synth and know which notes are being used. This could be expanded to see highlights in sequencer view too (via audition pads). Just putting it out there for discussion for another time!

### Is there a relevant Pull request?

#389

### What hardware did you reproduce it with?

OLED, 7SEG

### What firmware did you reproduce it with?

nightly

### What is the firmware name:

2023_09_05-593bfc5

### If possible provide the steps to reproduce the issue and upload additional media:

1) Make sure Highlight Incoming Notes is ON in the community features menu
2) Enter a Synth, MIDI, or CV clip
3) Either learn midi keyboard to clip or switch MIDI clip to the corresponding channel of midi keyboard.
4) Play notes and observe highlight behavior.

soymonitus, 2023-09-07:
My vote is for white pads with brightness independent of velocity, or at least with a linear curve that has the starting brightness (for the lowest velocity) be a bit more brighter, so low velocity can also be seen

PaulFreund, 2023-09-11:
My suggestion would be to ignore velocity and highlight with the same intensity as if a user presses a keyboard pad. Any votes against? I'd also enable this by default.