SynthstromAudible/DelugeFirmware #530: Change Automation Community Features settings to defaults + de-activate for audio clip

seangoodvibes, 2023-09-08:
### Please describe the problem:

Per guidance from @m-m-adams:

1. Move Automation Interpolation default setting to the Defaults Menu
2. Disable Clear Clip/Shift Clip automation changes on Audio Clip
3. Make Clear Clip/Shift Clip/Nudge Note automation changes permanent for Instrument Clip

### What is the expected behavior?

No community features menu required for automation view

Make use of defaults menu where possible

### Is there a relevant Pull request?

https://github.com/SynthstromAudible/DelugeFirmware/pull/241

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Built from release/1.0

seangoodvibes, 2023-11-03:
@m-m-adams based on the question we got on discord today, I think we need to re-consider the following proposed change:

- Make Clear Clip/Shift Clip/Nudge Note automation changes permanent for Instrument Clip

I think some users will want to adjust automation together with their notes. E.g. the user today asked how to shift/nudge automation together with their notes.

Should we just move all the automation view community features settings to defaults?

m-m-adams, 2023-11-09:
Yeah maybe it should be a 'link' option, where shifting automation shifts notes and vice versa

closed 2024-01-28 by seangoodvibes