SynthstromAudible/DelugeFirmware #531: Linked Automation-View of one clip to another clips Arp-Rate bug

bnhrsch, 2023-09-09:
### Please describe the problem:

I somehow managed to "link" the automation of a Synth/Midi-Clip to the Arp-Rate of a completely separate Kit-Clip. 

Automating a value in this particular Synth-/Midi-Clip done via the Automation-View will cause the Arp-Rate of the "linked" Kit-Clip to fluctuate. The values of the Arp-Rate seem to be consistent to what is drawn in the other clips Automation-Lane. 

This does not affect other channels besides these two. If the Midi-Clip is muted, or the automation deleted, the ratcheting stops. Changing the Synth-Clip to a Midi-Clip or vice versa does not change this behavior. 

I was unable to reproduce this effect with a new song but here is the file with the two linked clips. It only has the two clips- I deleted everything else from the file.  [https://we.tl/t-WZlNRgeyC5](https://we.tl/t-WZlNRgeyC5)

Clip-1 is a Midi-Clip. In it automation of it's HPF-Freq is drawn via the Automation-View
Clip-2 is the "linked" Kit-Clip with one row of an 808 with activated Arpeggiator.
Starting the song with both Clips unmuted will cause a ratcheting effect of the 808-Arp.
Muting the  Midi-Channel or deleting the Automation will stop the ratcheting. 

I was unable to reproduce the Bug since it happened during production and was discovered pretty late. I suspect that copying and changing a Clip from Midi- to Synth or Kit might have caused the bug. 

### What is the expected behavior?

Automation drawn in the Midi-Clip should not affect the Arp-Rate of another clip. 

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

nightly

### What is the firmware name:

deluge-v1_0_0-nightly+2023_09_09-799dc8b.bin

seangoodvibes, 2023-09-10:
@bnhrsch are you able to confirm if this bug exists with automation set the original way (outside automation view)

Need to confirm if this is an automation view bug or a core automation bug.

Thanks 

Sean

bobtwinkles, 2023-12-17:
This might be related to the fix in #841  -- if the param collection memory was left pointing at something left behind by the other clip, we could get this behavior.

closed 2024-03-03 by m-m-adams