SynthstromAudible/DelugeFirmware #538: Grainfx uses significant amounts of DSP even when muted

m-m-adams, 2023-09-12:
### Please describe the problem:

Each grainfx buffer allocates a 524kB buffer and adds 20% runtime to the audio engine render function (measured with the audioLog timestamps on a sample song provided to us). This slowdown occurs regardless of whether the track with the grain is active or muted.

Suggested fixes:
1. Treat grain FX like reverb (i.e. one global grain buffer/render)
2. Shorten grain buffer
3. vectorize grain rendering

### What is the expected behavior?

Less slowdown, performance recovers with muting

### Is there a relevant Pull request?

#363

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

nightly

m-m-adams, 2023-09-14:
Still an issue with 540 - if the audio clip with the grainfx is culled, the dsp load isn't actually lightened

PaulFreund, 2023-10-23:
Not sure where, but fixed

closed 2023-10-23 by PaulFreund