SynthstromAudible/DelugeFirmware #547: Align automation view with deluge UX

m-m-adams, 2023-09-14:
### Please describe the problem:

Automation view has some spots where the UX does not align with the rest of the deluge. In order to match behaviour and muscle memory with existing features I think the following should be changed:

1. Parameter selection should be using the editing menu structure rather than a single list of parameters. The menu should open on click, and the select encoder should adjust the value. Selecting a parameter with a shortcut serves as quick access and the menu structure aids in discoverability for people who aren't familiar with the shortcuts.
2. Clicking select should open the editor menu (like in clip view)
4. Shift+click select should open the global menu (or change everywhere else to open view specific menus instead - the confusing UX is having this one view that has no access to the global menu. Maybe open the global menu paged to the automation menu) 
5. Turning select should edit the currently selected parameter (like in clip view)
6. Gold knobs should work as gold knobs for fast editing of synth parameters. The current behaviour for the keyframe mode could be locked to custom 2/3 and free up the rest of the buttons. This allows tweaking other parameters while automating one, which is very useful for interdependent things like wave position and filter cutoff. 
7. Changing interpolation affects future actions and not the current state of interpolation, so I don't think it should be on a toggle. Maybe hold shift while pressing the pads or something? Right now there is a hidden state (neither visible nor audible) which changes interactions and that is very much not ideal
8. Audition pads + shortcut pad should function the same as shift+shortcut (#546). For kits this is very useful to always know exactly which drum you're targeting instead of thinking about what you've selected. For synths/midi it really just matters to be consistent with kits
9. The automation overview is a solid addition that would be nice to have in the clip view (highlighting modulated parameters instead of automated ones)

Generally follow these guidelines: https://github.com/SynthstromAudible/DelugeFirmware/blob/community/docs/ux_principles.md 

If there is a guideline that we believe isn't worth following we should modify those principles but I believe they had pretty wide approval when written

### What is the expected behavior?

For behaviours which are common between clip and automation view the shortcuts and interactions should be the same. Having two extremely visually similar views with very different interaction is confusing and hurts the ability to quickly switch between them

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

nightly

trappar, 2023-09-14:
Hard disagree on number 5, and I think the current UX is actually aligned with Deluge's other UX. Deluge's current UX only ever gives the ability to view parameters at currently visible levels of depth.

When in song view, you see multiple clips and if you edit parameters for one of them, but only if you manually select it. Without that manual selection there is no context for what you would be editing. Also, when editing parameters, you're always doing so with "affect entire" enabled because you are viewing at a level that only gives you information about clips as a whole. There is no access to row level parameters.

When you jump to clips, you can now edit all via affect entire, or in the case of a kit, rows individually. This is because you're viewing at a level that gives information about just one clip and its rows.

When you jump to automation view, you are now only looking at one parameter. There is nothing outside of that parameter visible. The mod encoders no longer make sense in their original context. I would actually suggest that the quick selection buttons (for switching the mod encoders to different mappings) and the affect entire button make no sense within this view and should be disabled similar to how parameter editing is disabled in song mode when no track is selected.

sichtbeton, 2023-09-14:
I agree with 

> ### Please describe the problem:
> Automation view has some spots where the UX does not align with the rest of the deluge. In order to match behaviour and muscle memory with existing features I think the following should be changed:
> 
> 1. Parameter selection should be using the editing menu structure rather than a single list of parameters. The menu should open on click, and the select encoder should adjust the value. Selecting a parameter with a shortcut serves as quick access and the menu structure aids in discoverability for people who aren't familiar with the shortcuts.
> 2. Clicking select should open the editor menu (like in clip view)
> 3. Shift+click select should open the global menu (or change everywhere else to open view specific menus instead - the confusing UX is having this one view that has no access to the global menu. Maybe open the global menu paged to the automation menu)
> 4. Turning select should edit the currently selected parameter (like in clip view)
> 5. Gold knobs should work as gold knobs for fast editing of synth parameters. The current behaviour for the keyframe mode could be locked to custom 2/3 and free up the rest of the buttons. This allows tweaking other parameters while automating one, which is very useful for interdependent things like wave position and filter cutoff.
> 6. Changing interpolation affects future actions and not the current state of interpolation, so I don't think it should be on a toggle. Maybe hold shift while pressing the pads or something? Right now there is a hidden state (neither visible nor audible) which changes interactions and that is very much not ideal
> 7. Audition pads + shortcut pad should function the same as shift+shortcut ([Automation view : add back audition + shortcut combo #546](https://github.com/SynthstromAudible/DelugeFirmware/issues/546)). For kits this is very useful to always know exactly which drum you're targeting instead of thinking about what you've selected. For synths/midi it really just matters to be consistent with kits
> 8. The automation overview is a solid addition that would be nice to have in the clip view (highlighting modulated parameters instead of automated ones)
> 
> Generally follow these guidelines: https://github.com/SynthstromAudible/DelugeFirmware/blob/community/docs/ux_principles.md
> 
> If there is a guideline that we believe isn't worth following we should modify those principles but I believe they had pretty wide approval when written
> 
> ### What is the expected behavior?
> For behaviours which are common between clip and automation view the shortcuts and interactions should be the same. Having two extremely visually similar views with very different interaction is confusing and hurts the ability to quickly switch between them
> 
> ### Is there a relevant Pull request?
> _No response_
> 
> ### What hardware did you reproduce it with?
> OLED
> 
> ### What firmware did you reproduce it with?
> nightly
> 
> ### What is the firmware name:
> _No response_
> 
> ### If possible provide the steps to reproduce the issue and upload additional media:
> _No response_

Agree entirely. 

Also to 5: the way the gold knobs are learned to paramters in a kit row changes from every row. this should be reflected in the automation view (be consistent). if you press affect entire, the usual kit mappings should be present (does so afaik).

when in automation view and you are looking at lets say lpf but your gold knob is mapped to hpf, when you turn it it should jump to hpf and automate it. if no gold knob is learned to your current parameter, you can enter values either on the grid or via turning select (following Deluge UX).

sichtbeton, 2023-09-14:
> ### Please describe the problem:
> Automation view has some spots where the UX does not align with the rest of the deluge. In order to match behaviour and muscle memory with existing features I think [...]

Also very much agree on 8.) having this overview in clip mode would be awesome and helpful.

m-m-adams, 2023-09-14:
> Hard disagree on number 5, and I think the current UX is actually aligned with Deluge's other UX. Deluge's current UX only ever gives the ability to edit parameters at the currently visible level of depth or through some additional context.
> 
> When in song view, you see multiple clips and cannot edit parameters for any of them unless you manually select a clip. Without that manual selection there is no context for what you would be editing. Also, when editing parameters, you're always doing so with "affect entire" enabled because you are viewing at a level that only gives you information about clips as a whole.
> 
> When you jump to clips, you can now edit all via affect entire, or in the case of a kit, rows individually. This is because you're viewing at a level that gives information about just one clip and its rows.
> 
> When you jump to automation view, you are now only looking at one parameter. There is nothing outside of that parameter visible. The mod encoders no longer make sense in their original context. I would actually suggest that the quick selection buttons (for switching the mod encoders to different mappings) and the affect entire button make no sense within this view and should be disabled similar to how parameter editing is disabled in song mode when no track is selected.

The mod encoders still make sense. You're obviously not just looking at a single parameter, it's a whole synth (see the audition pads still playing notes from the underlying synth). Not being able to adjust other parameters quickly is incredibly annoying, and the view only needs two knobs to work. There is literally nothing lost by putting the current behaviour on custom 2/3 and restoring the original behaviour on every other gold button

trappar, 2023-09-14:
> The mod encoders still make sense. You're obviously not just looking at a single parameter, it's a whole synth (see the audition pads still playing notes from the underlying synth).

In this view, the grid and display only show values for a single parameter over time. The fact that you can audition doesn't change that. Look at arrange view for similar UX. You can audition in that view, and yet it doesn't give access to mod encoders without some additional context.

You do make a point though. Interacting with the audition pads is additional context that you're intending to work with the whole instrument rather than just the parameter currently viewed. Maybe it should re-enable the clip version of mod controls specifically when holding an audition pad?

> Not being able to adjust other parameters quickly is incredibly annoying

I disagree. I've worked with it a lot and I find it super quick to jump out of the view or jump to a specific parameter. When I use automation view it's because I specifically want to modify automation. If I wanted to sculpt the overall sound, I would be in clip view.

This is also not entirely relevant to the conversation. It's perfectly valid for you to be annoyed with the feature and want to change it in a way that's less annoying to you, but you are posing this as inconsistent with the rest of the UX and I'm simply stating why I don't think that's true.

> There is literally nothing lost by putting the current behaviour on custom 2/3 and restoring the original behaviour on every other gold button

There's plenty that would be lost due to that change:
1. Hides relevant controls for this view in a place that's virtually un-discoverable. This is especially concerning for selection mode.
2. Adds additional button press to access relevant controls for this view, and then even more button presses to switch back when leaving the view.
3. Makes UX for custom 2/3 inconsistent with the rest of Deluge.

m-m-adams, 2023-09-14:
Great that you're used to it but I don't find it easy. Only overriding 2 gold knobs instead of 16 takes nothing away from you and makes the mod encoders work more consistently. Yes custom 2/3 should be selected when you enter automation view. This change just leaves 

You can continue to say that automation view is separate from clip view but they are very obviously closely linked in that they are the same clip. For example you can record sequencer data from automation view, and you can record automation data from the clip view. Keeping as much as possible consistent between the two views is important.

The gold knobs are functional in a consistent way in every view of the deluge, including song and arranger view... they only work in affect entire in those modes because there is no non-affect entire target.

sichtbeton, 2023-09-15:
> Great that you're used to it but I don't find it easy. Only overriding 2 gold knobs instead of 16 takes nothing away from you and makes the mod encoders work more consistently. Yes custom 2/3 should be selected when you enter automation view. This change just leaves
> 
> You can continue to say that automation view is separate from clip view but they are very obviously closely linked in that they are the same clip. For example you can record sequencer data from automation view, and you can record automation data from the clip view. Keeping as much as possible consistent between the two views is important.
> 
> The gold knobs are functional in a consistent way in every view of the deluge, including song and arranger view... they only work in affect entire in those modes because there is no non-affect entire target. If you hold an audition pad for a row then they work normally.

Yep. The two methods don't need to fight but can co-exist beautifully.

And its way more consistent. It truly is. The gold knobs and the selection work almost everywhere the same. no reason to differ here. really none.

sichtbeton, 2023-09-15:
> There is literally nothing lost by putting the current behaviour on custom 2/3 and restoring the original behaviour on every other gold button
> There's plenty that would be lost due to that change:
> 
> Hides relevant controls for this view in a place that's virtually un-discoverable. This is especially concerning for selection mode.

Nothing is lost. Does not make sense tbh. 

> Adds additional button press to access relevant controls for this view, and then even more button presses to switch back when leaving the view.

It doesn't. Why?

> Makes UX for custom 2/3 inconsistent with the rest of Deluge.

If anything adding the standard gold knob functions back in, it moves the audition view much closer to the original UX. 

> If custom 2/3 was automatically selected when entering automation view then it would help with the first two points, but I could also see that being annoying/unexpected.

Right now, having  basically two useless gold knobs (do the same) and all the selection encoders rendered useless too is unexpected (and needlessly so).

So you like the function, nice. But you truly have no argument here. Sorry.

sichtbeton, 2023-09-15:
To the last point, it could work like this:

- change to the automation view via clip
- select a gold knob via gold knob selections
- turn one of them
- automation lane of this parameter is shown

sichtbeton, 2023-09-15:
All in all: i agree fully with all the points that @m-m-adams makes. The automation view feels rather foreign to the deluge and would (in my opinion) benefit highly from being a smoothly integrated part of it. It just makes for a more streamlined and enjoyable machine in my view.

closed 2024-01-08 by m-m-adams