SynthstromAudible/DelugeFirmware #555: MPE instrument sending to other learned channels while recording

alifeinbinary, 2023-09-18:
### Please describe the problem:

While recording, the Deluge is sending MIDI notes and CC data received from a MPE device connected to the USB host port to other MIDI devices connected via DIN MIDI that have been learned. It's like `doingMidiThru = true` while recording.

Here's how I have things configured, just to give you a quick idea how I have things set up.

### MIDI devices connected to DIN MIDI via a MIDI hub
Nord - Channels 1 through 4
Elektron Keys - Channels 5 through 8 
Elektron Rytm - Channels 13, 14
Oh-coast - Channel 16

### MPE device connected to USB host MIDI
Osmose > Roland UM-One > USB host

### Deluge settings
Differentiate inputs > Enabled
MIDI Thru > Disabled
MIDI > Devices > DIN = (in: upper > 0, lower > 0 | out: upper > 0, lower > 0)
MIDI > Devices > UM-ONE = (in: upper > 14, lower > 0 | out: upper > 14, lower > 0)

### What is the expected behavior?

It's expected that the MIDI recorded from a MPE device on the USB host port isn't sent to other MIDI devices connected to the Deluge's DIN ports even when there's channel overlap. 

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Built from community

### What is the firmware name:

1.0.0-DEV-664C410

### If possible provide the steps to reproduce the issue and upload additional media:

1. Turn on
2. Apply the following settings to the Deluge 

- Differentiate inputs > Enabled
- MIDI > MIDI Thru > Disabled
- MIDI > Devices > DIN = (in: upper > 0, lower > 0 | out: upper > 0, lower > 0)
- MIDI > Devices > UM-ONE = (in: upper > 14, lower > 0 | out: upper > 14, lower > 0) or whatever your MPE synth requires

3. Learn MIDI synth connected to DIN ports
4. Create new track, learn MPE synth to it
5. Enable record, press play
6. Play the MPE synth long enough to hear notes being played on the synth connected over DIN MIDI when there's channel overlap between what the MIDI synth receives on and the MPE synth is sending

m-m-adams, 2023-09-18:
I'm not able to reproduce this using a very similar setup so I have a few follow up questions:

1. Are you able to set up a midi monitor to see which specific channels are being sent to the wrong midi output? If not which synths are getting triggered by overlapping channels? 

2. Does the bug still occur with differentiate inputs off? (it's not necessary for this setup)

3. Does this setup work on the official firmware?

m-m-adams, 2023-09-28:
I can't get notes to come through but I can get aftertouch and pitch bend through - is that what you meant?

closed 2023-10-08 by m-m-adams