SynthstromAudible/DelugeFirmware #564: Freeze of OLED screen and all pads that are connected to the PIC

PaulFreund, 2023-09-22:
### Please describe the problem:

After some runtime (depending on situation) the interrupts for 

DMA_INTERRUPT_0 + 10 (PIC_TX_DMA_CHANNEL)
DMA_INTERRUPT_0 + 11 (PIC_RX_DMA_CHANNEL)

are no longer fired, severing all interaction with OLED and pads. At least for me for some time I could still start and stop playing

### What is the expected behavior?

This does not happen

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

nightly, Built from release/1.0, other

### What is the firmware name:

Traced back to at least e3d595693595c615cb01f1fd06e424fc240f79b8

### If possible provide the steps to reproduce the issue and upload additional media:

1. Load a song (relevant parameters unknown)
2. Open clip view of a playing clip to see playhead scrolling through
3. Set tempo to FAST (max)
4. Wait for a random amount of time (shortest for me was 1 min, highest was >1hour without happening)

Happens in debug and release, in debug you can see that the interrupts above are no longer called

PaulFreund, 2023-09-23:
[Hangtest2.zip](https://github.com/SynthstromAudible/DelugeFirmware/files/12706773/Hangtest2.zip)

This is basically the song I replicated with but with a different kit because of sample origin

stellar-aria, 2023-10-13:
Tentatively closing  as fixed by #611, if we find this is still present the issue can be reopened.

PR #611 (2023-10-13): Incorporated changes according to spec from recent Renesas RZA1LU_Bli…
Fixes #564,

Changes:
* Updated interrupt controller interrupt handler assembly and user handler function according to Renesas RZA1LU_BlinkySample. Update: After doing this correctly it seems to solve the problem
* Make sure cache is flushed before DMA is accessing data
* Prevent routineForSD to run from interrupt to prevent possible unlocked data access
* Disabled AUDIO_LOG
* Original Problem can be detected by checking if INTC.ICCRPR != 255 (equalling idle priority)

see also https://github.com/SynthstromAudible/DelugeFir [... cut]
closed 2023-10-13 by stellar-aria