SynthstromAudible/DelugeFirmware #568: Naming of tracks not possible in arranger anymore

sichtbeton, 2023-09-23:
### Please describe the problem:

In arranger when holding audition + name the keyboard comes up. Usually you could change the name of  the track to whatever you like. 

But you can’t type anymore, the name is stuck.

### What is the expected behavior?

Being able to type with keyboard

### Is there a relevant Pull request?

Not that I know of

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

nightly

### What is the firmware name:

1.0.0 dev 07057CF

### If possible provide the steps to reproduce the issue and upload additional media:

- open arranger with SONG until it’s blinking
- audition button + “name”
- Try to type with keyboard

m-m-adams, 2023-09-24:
This is still working for me

sichtbeton, 2023-09-24:
> This is still working for me

strange. same fw?

m-m-adams, 2023-09-25:
Yeah, working on latest nightly and the PRs I've reviewed today

sichtbeton, 2023-09-25:
alright, will close this. maybe some weird bug.

closed 2023-09-25 by sichtbeton