SynthstromAudible/DelugeFirmware #574: Polyphonic Kit row will cut itself off via Velocity

ok-reza, 2023-09-26:
### Please describe the problem:

If a long polyphonic sample (set to once) is sequenced at velocity 65 then another instance of the sample is sequenced right after at velocity 5, the velocity 65 sample gets cut off by the velocity 5.

### What is the expected behavior?

Both the velocity 65 and velocity 5 should play out completely. This correct behavior is seen in nightly 09/09 (I haven't checked more recent nightlies) and the current official firmware.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

nightly

### What is the firmware name:

2023_09_26-3b29423

### If possible provide the steps to reproduce the issue and upload additional media:

1) Create a new kit.
2) Load a long sample.
3) Set POLYPHONY to POLYPHONIC and SAMPLE MODE to ONCE
4) Sequence a note at the first pad of a 1 bar loop.
5) Add another note shortly after, lets say on beat 2 of the 1 bar loop.
6) Adjust velocity to 2nd note to 5.
7) Upon playing, you will hear the 1st note either cut out when the 2nd note plays, or match the velocity of the new note as it plays out (it's hard to tell which of the two behaviors it's doing)

Alternatively, if you switch to Kit Keyboard View, you may be able to notice this simply by pressing different velocities of the same sample in quick succession.

m-m-adams, 2023-09-26:
Isn't this how it's always worked? You've never been able to play the same drum row twice at the same time

Were oneshot samples special?

ok-reza, 2023-09-26:
This is an issue when both ONCE _and_ POLYPHONIC are selected for the kit row, in which case you are normally able to hear multiple instances of the sample getting triggered.

m-m-adams, 2023-09-28:
Can't reproduce - followed the instructions and both samples play fully

closed 2023-09-28 by m-m-adams