SynthstromAudible/DelugeFirmware #580: ODR violations

m-m-adams, 2023-09-28:
### Please describe the problem:

There are piles of ODR violations since merging #567 

### What is the expected behavior?

Not having them

### Is there a relevant Pull request?

#567

### What hardware did you reproduce it with?

OLED, 7SEG

### What firmware did you reproduce it with?

Built from community, Built from release/1.0

duncanrmi, 2023-11-13:
what is ODR?

bobtwinkles, 2023-11-14:
ODR is the [one definition rule](https://en.cppreference.com/w/cpp/language/definition). This bug is for some internal/technical problem, we haven't observed any user-visible problems associated with this (yet..)

closed 2024-03-03 by m-m-adams