SynthstromAudible/DelugeFirmware #582: Synth preset drum rows cause duplications when switching between Instrument types

stellar-aria, 2023-09-29:
### Please describe the problem:

When creating a kit using a synth preset and switching to another instrument type and back, the kit rows remain the same, as per standard behavior with kits.

### What is the expected behavior?

No new rows are created, as per standard behavior, and the only row that exists is the synth preset.

Also, that E073/S001 does not popup, and that the correct initial Synth Preset number (i.e. 0) is displayed.

Compare this to creating a new empty Kit (using SHIFT + [KIT]) and switching to [SYNTH] and back.

### Is there a relevant Pull request?

#395

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

nightly

### What is the firmware name:

1.0.0-nightly-C56CDB5

### If possible provide the steps to reproduce the issue and upload additional media:

- Boot Deluge
- Press SHIFT + [KIT] to create a new kit
- Press BACK to back out to empty kit
- Press the sole Kit row's [AUDITION] + [SYNTH] to load a synth preset (desired preset 0)
  - At this point E073 pops up onto the display, and when it disappears there's no preset number (and the sound generated is an empty synth patch) until scrolled at least once, at which point "1" will show up instead of the actual first preset, "0".
  - Occasionally, S001 will show up and freeze the Deluge
- Press [SELECT] to select a synth preset
- Verify that the Synth-preset Kit row's [AUDITION] pad generates the correct sound
- Press [SYNTH]
- Press [KIT]
  - E073 pops up again
- There are now three Kit rows:
  - The topmost is SYNT000.
  - The second has no name when pressed but still creates a sound
  - The bottom-most row (the one that should have been set to a synth preset) is now empty once again

m-m-adams, 2023-10-03:
Is this still happening with 577? I haven't been able to trigger it recently

If so could you share your first preset? I have a theory that it's related to param managers failing to load. I added S001 while trying to track it down, it's caused by a null de reference somewhere

m-m-adams, 2023-10-03:
Ah I just tried on 7seg and got the error you mentioned. There must be a difference in how their browsers are initialized

PR #588 (2023-10-04): Bugfix/synth kit e073 s001
Fixes a null pointer de reference if not replacing an instrument

Fixes the first load getting over written by default drum

Fixes lingering drums in memory without note rows

Fix #582 

I believe there is a memory leak in the sound->readFromFile method and memory gets full after ~300 preset loads. That's not fixed by this PR but I'll hopefully get it with expanded cpputest coverage
closed 2023-10-04 by bobtwinkles