SynthstromAudible/DelugeFirmware #591: Stale version string in local builds

bobtwinkles, 2023-10-03:
### Please describe the problem:

Version strings from the same local build are stale if the build directory is not deleted (or CMake used to manually re-run the configure step).

### What is the expected behavior?

https://github.com/SynthstromAudible/DelugeFirmware/blob/f3b2407134d76a811a0f01b8e7c15d28b1792f90/src/deluge/CMakeLists.txt#L1 should use https://cmake.org/cmake/help/latest/command/file.html#generate instead, so it reruns on every *build* not every *configure*.

### Is there a relevant Pull request?

N/A

### What hardware did you reproduce it with?

OLED, 7SEG

### What firmware did you reproduce it with?

nightly, community, Built from community, Built from release/1.0

stellar-aria, 2023-10-14:
Generate doesn't do variable replacement, as far as I'm aware

m-m-adams, 2023-10-14:
Maybe we could have dbt build release always run the configure step?

m-m-adams, 2023-11-12:
What if we just make a commithash.h file and set the content with dbt or git hooks?

closed 2024-03-03 by m-m-adams