SynthstromAudible/DelugeFirmware #593: Modfx parameter got modulated at the same time as the filter cutoff

soymonitus, 2023-10-04:
### Please describe the problem:

Some modfx parameter was tied to a LFO while I was adjusting the LFO amount for the filter cutoff. I was adding some S&H modulation to the filter cutoff from a fresh new synth (Shift + Synth). Then i went to change modfx type (selecting Stereo chorus) and the sound was super weird, as if the rate of the stereo chorus was modulated also by the S&H lfo. Don't know if there will be a bug or not but maybe someone can check if there is some code path that will allow several parameters to be modulated at the same time.

### What is the expected behavior?

If I add modulation for the filter cutoff, only the filter cutoff is modulated and not other parameters at the same time

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

nightly

### What is the firmware name:

deluge-v1_0_0-beta+2023_10_02-d9fbde6

### If possible provide the steps to reproduce the issue and upload additional media:

Not able to reproduce now

m-m-adams, 2023-10-04:
Confirmed, some mod routings seem to go to the wrong spot. Modulating env 2 modulates env 1, modulating vibrato modulates transpose, osc 2 hits osc 1 and probably more

soymonitus, 2023-10-21:
From your description it seems they are  all just one position off, like someone introduced a new element in an array and forgot to increase the indexing

soymonitus, 2023-10-21:
I was going to pick this bug to fix it but can't reproduce it now. Was this fixed as part of another PR?

m-m-adams, 2023-10-22:
I've still got it showing up when modulating osc 2, mod 2 and env 2

PR #723 (2023-11-17): Bugfix: handle case of indexed params
Fix #593 

In the code for the mod matrix a check was lost to make sure we highlight the correct version of indexed menus (e.g. osc, env, lfo)
closed 2023-11-17 by PaulFreund