SynthstromAudible/DelugeFirmware #596: Delay sync - gold knob clicks and menu disagree with each other

colinmeyer, 2023-10-06:
### Please describe the problem:

The note-level of sync reported by clicking the lower golden knob or by shift-delay-sync disagree

E.g. fresh kit clip, select any row with audition pad. Click lower golden knob once, dispalys flashes "64th". Shift-delay-sync, display shows "32nd".

It is possible to get the golden knob behavior into a funky state by clicking the top knob while the sync value is open in the menu (or shortcut) - both golden buttons cause display to flash "dele"

### What is the expected behavior?

Two values should be the same
Also expect "off" as a sync level to cycle through, while clicking lower gold knob

### Is there a relevant Pull request?

282

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

community

### What is the firmware name:

deluge-v1_0_0-beta+2023_10_02-d9fbde6.bin

### If possible provide the steps to reproduce the issue and upload additional media:

1. turn on
2. make sure community feature for delay is on
3. new song
4. press kit
5. press bottom audition pad
6. select delay time/amount button between gold knobs
7. click lower gold knob - display flashes "64th"
8. shift-delay-sync - dispåy shows "32nd"

soymonitus, 2023-10-08:
At least on my OLED deluge the changes on the golden knobs get reflected on the selected delay param and even if I change the delay sync with the select knob, then pressing the golden knobs continues cycling to the next value correctly. What is it exactly what is happening to you?

https://github.com/SynthstromAudible/DelugeFirmware/assets/6472762/158ea332-62c7-46a3-b23f-73a2e7327cff

colinmeyer, 2023-10-09:
Just like I say in the repro steps: clicking lower knob reports "64th" but when I go in and check via menu, it shows "32nd"

https://photos.google.com/share/AF1QipMV3S3WpYQGq2uC3ep_umrCJvCq0aT9fgZu7D30cG4sheSgocXsRWkGs8FakyY3fg?key=THJtSFl3T2FULXpGQ1hBd3BnY1BHZDZ3ZFRMdDFn

colinmeyer, 2023-10-09:
Separate, but probably related bug: either top or bottom gold knob report "dele" and don't do anything when clicked, as can be seen in the above linked video

colinmeyer, 2023-10-11:
Made another video that better illustrates the issue
https://photos.app.goo.gl/95iYCZpDJqxAqXtg6

m-m-adams, 2023-10-14:
Confirmed, appears to be off by one in the sync list

m-m-adams, 2023-10-14:
Problem seems to be that delay sync actually goes from 2-bar to 128th notes now. I'm not sure why/how that changed

closed 2023-10-22 by m-m-adams