SynthstromAudible/DelugeFirmware #600: Switching from synth/midi to kit irreversably transposes notes to kit notes

o010b, 2023-10-08:
### Please describe the problem:

If (accidentally) switching the instrument from synth/midi to kit, all current note data is transposed to fit the notes of the kit. Switching back to synth/midi retains the transposed notes. Undo does not work. 

### What is the expected behavior?

Switching back to original note pitches. 

Alternatively, press undo button to reverse transformations.

Alternatively, a warning/confirmation before switching a track mode if the track contains note data. 

Maybe all three.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

community

### What is the firmware name:

4.1.3

### If possible provide the steps to reproduce the issue and upload additional media:

1. Create a glorious synth track. Wear sunglasses at night. 
2. Butter fingers and excitedly flub the kit button.
3. Sigh of the disappointment of your being and switch back to synth mode.
4. Enjoy experimental atonal cacophony that surely will revolutionize the world of music.

m-m-adams, 2023-10-08:
What happens on the official firmware?

o010b, 2023-10-10:
My bad, it is the official firmware, not community edition.