SynthstromAudible/DelugeFirmware #606: Audio recording and resampling doesn't work

Lugaseurop, 2023-10-09:
### Please describe the problem:

When I go to record audio on an audio track, it starts to record as normal and then a few seconds in, the record button stops flashing and the track reacts like I've hit the record button (I haven't) and discontinues the recording. The same happens for resampling. 

### What is the expected behavior?

The audio should continue to record until i press the correct button to make it stop. 

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

community

### What is the firmware name:

Deluge Beta

m-m-adams, 2023-10-09:
I can't reproduce - audio recording is working on my deluge. Can you provide more info on exactly what you're doing?

Lugaseurop, 2023-10-09:
I was rerecording the audio formt he deluge output (post fx) in the audio channel. I  have since tried again and it all seems to be working but I had the problem consistenly over a couple of days! Will try to reproduce tomorrow and provide more details.

closed 2023-10-13 by m-m-adams