SynthstromAudible/DelugeFirmware #610: Scrolling through synths - sample based synths don't always load.

Radius9999, 2023-10-11:
### Please describe the problem:

When choosing a synth for a track, sample based synths don't always load.  Sometimes I have to go past it and then scroll back to get it to load.  This is on OLED with community build beta.  It's pretty consistent in not loading most of the time.

### What is the expected behavior?

Sample based synths should load just like regular synths.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

community

### What is the firmware name:

deluge-v1_0_0-beta+2023_10_02-d9fbde6.bin

### If possible provide the steps to reproduce the issue and upload additional media:

Use scroll knob to browse various saved synths on a track.  Name will show, but no response to trigger input via midi keyboard or audition buttons.  Scroll a couple of synths past and then back again usually fixes it.  Mostly going past and back.  Going before and back usually doesn't work.

davidvandenbor, 2023-10-11:
Just a small chime in. I've noticed the same bug. I've also noticed that the firmware versions of the last couple of months are riddled with bugs, making the Deluge at times completely unplayable: wondering if the firmware is starting to hit the **hardware limit** of the device. Meaning: We can want the Deluge to do lots of amazing things, but if the downside results in an unusable device, is it really worth it? At a certain point, the **hardware** will say NO! 😃 At the moment, I'm back at the firmware version that came out right before the **Grain FX** merge of @alter-alter 

Have to say though: the community have added some pretty amazing improvements to the device software wise!

soymonitus, 2023-10-12:
> Just a small chime in. I've noticed the same bug. I've also noticed that the firmware versions of the last couple of months are riddled with bugs, making the Deluge at times completely unplayable: [...]

@davidvandenbor What was different before Grain FX commit? Is there a noticeable performance difference between current build and that point in the past?

Radius9999, 2023-10-13:
> At the moment, I'm back at the firmware version that came out right before the **Grain FX** merge of @alter-alter

Where can I find the older builds?  Is there a listing somewhere?  There are only two new features that I really like/need.  The sticky shift (brilliant!) and the stereo chorus.  Oh, and the audio recording beginning edit without changing speed.  Everything else I could do without.

m-m-adams, 2023-10-13:
> Just a small chime in. I've noticed the same bug. I've also noticed that the firmware versions of the last couple of months are riddled with bugs, making the Deluge at times completely unplayable: [...]

We are limited by the hardware but the open issues are just software bugs, they should be fixable. If you could open reports on issues you're having it would be a big help!

m-m-adams, 2023-10-13:
Tentatively fixed - please re open if you still experience this

Radius9999, 2023-10-15:
Fixed on what version?  I tried deluge-v1_0_0-beta+2023_10_14-79bea14 and it's still happening.  What I figured out was that it's not that it's not loading, it's that it's loading very slowly, and on multi-samples if you hit a key before it finishes loading is stops loading.  So multi-samples will only load say half the keyboard and then the other samples don't load.  It would be good if the "loading" message stayed on the screen until all the samples were loaded.  Also with single sample synths, if it's a long sample and you hit a key, it will play a portion of the sample and stop, and then if you hit a key again it will play it longer and stop, etc, until it finishes loading.  Something is causing samples to load very slowly, thus giving the perception that they're not loading at all.

Radius9999, 2023-10-15:
> Tentatively fixed - please re open if you still experience this

Fixed on what version? I tried deluge-v1_0_0-beta+2023_10_14-79bea14 and it's still happening. What I figured out was that it's not that it's not loading, it's that it's loading very slowly, and on multi-samples if you hit a key before it finishes loading is stops loading. So multi-samples will only load say half the keyboard and then the other samples don't load. It would be good if the "loading" message stayed on the screen until all the samples were loaded. Also with single sample synths, if it's a long sample and you hit a key, it will play a portion of the sample and stop, and then if you hit a key again it will play it longer and stop, etc, until it finishes loading. Something is causing samples to load very slowly, thus giving the perception that they're not loading at all.

m-m-adams, 2023-10-18:
Reopened - it should have been fixed by #618 but I might have been waiting longer for them to load. Might be fully fixed now with #627

Radius9999, 2023-10-18:
Yes, I can confirm that it works now.  Thanks.

PR #618 (2023-10-13): Bugfix/remove sneaky optimization and replace with virtual functions
The sneaky optimization seems to have resulted in issues around multi samples (which I think is actually all samples internally)

With this fix I can no longer reproduce the following bugs:

Fix #610 
Fix #492
~~Fix #472~~ still happens sometimes
closed 2023-10-19 by m-m-adams