SynthstromAudible/DelugeFirmware #615: MIDI CCs learning don’t work correctly in old songs

Icoustic, 2023-10-13:
### Please describe the problem:

MIDI CCs don’t work correctly for audio clips and synth clips in old songs. 
Learning happens but the CCs don’t change the parameter. (In kits it works normally.)
Have to re-save old song with current new firmware, then the CCs work as expected.

### What is the expected behavior?

CC learning should work as expected in old songs too, shouldn’t it? Without having to re-save song first

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

nightly

### What is the firmware name:

Nightly-B6AECCA

Icoustic, 2023-10-13:
Actually scratch the “works as expected part” - *some* audio clips and synths work and others don’t, I can’t see an obvious difference that would be a reasonable reason. Pretty strange. I’ll link the song xml soon

m-m-adams, 2023-10-13:
The official firmware has a significant number of midi handling bugs that lead incoming normal midi to be incorrectly treated as MPE zones and vice versa. Most likely what you're encountering is midi that was incorrectly learnt to the wrong channel and is now being correctly interpreted as not matching.

Once you share a broken song xml I'll be able to confirm

m-m-adams, 2023-10-14:
Closing for now - please reopen if you can provide a problem XML

closed 2023-10-14 by m-m-adams

Icoustic, 2023-10-14:
> Closing for now - please reopen if you can provide a problem XML

Yeah I will, just forgot to save with new CC learns and bring before leaving studio area where Deluge is currently staying

Icoustic, 2023-10-14:
> Closing for now - please reopen if you can provide a problem XML

Aha! I found the answer, it wasn’t that. I was learning to and moving CCs in clip clones that were muted and not the last one to play, and the CC only (correctly) moves in the clip clone that was last unmuted.
I didn’t know that! Nice 👍