SynthstromAudible/DelugeFirmware #656: Convert old SD cards to a proper file system

m-m-adams, 2023-10-28:
### Please describe the problem:

The deluge originally shipped with a file system using only file names and read them in what are called 'slots' in the deluge code. An example name is

SONG001a

This consists of a prefix (SONG/SYNT/KIT/SAMP), a number 001, and a subslot (a)

There is a significant amount of legacy code to properly categorize and handle these old files and do things such as skip subslots while browsing, and it complicates the new folder browsing system.

It would be helpful to provide a web or desktop application which would convert the old files to a folder structure such as songs/001/a to reduce the backwards compatibility requirements on browsing

This affects users who did not switch to the new factory image following the 4.0 update

### What is the expected behavior?

A tool could be provided to update old SD cards and remove legacy requirements

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

other

alex-reid, 2023-12-14:
I'm looking in to this as part of a little side project I'm working on. My use first case is to create a tool to migrate `< v4` SD cards to `v4` format (rename `SYNT001` -> `Sync Bass` and then update all references in songs XMLs to match). I'm creating a proof of concept in nodejs which I'll pop on github once its working for people to test if they so desire.

As I have (naturally) overcomplicated the solution, I'm parsing the XML files and doing things properly, so it shouldn't be too hard to modify my code to fix the problems raised in this issue.

My plan, post proof of concept, is to create a desktop app using Electron or Tauri or similar that people can use to update their SD cards.

When I first got my deluge I was running `3.1.5` so my only real legacy XML references are from that era of deluge.

What I would love to get are some examples of songs and instruments from older Deluge versions so that I can modify my business logic accordingly! If anyone can help me out, it would be greatly appreciated!!

alex-reid, 2023-12-14:
So after a bit of discussion, probably all that is necessary to achieve removing the legacy slot browsing code is remapping the song XML so that references to `presetSlot, presetSubSlot, instrumentPresetSlot, instrumentPresetSubSlot` reference the proper `presetName, presetFolder` structure of v4. I'll work off the following process:

**MVP**
Nodejs app to scan through all songs and remap params to v4 spec.

**v2**
Rename existing presets to v4 names `SYNT001` -> `001 Sync Bass` and `KIT003` -> `003 TR-909`

**v3**
Rewrite multiple song variants to subfolders `SONG123A.XML` -> `SONG123/A.XML` or `SONG/123/A.XML` depending on preference.

**v4**
Wrap the app with a simple GUI and provide it to users.

**Future plan**
If all goes well, I could envision this being a neat library manager for things like: 
- moving and renaming synths, kits and samples. 
- fixing files with broken references
- adding simple categories and tags to XML files that could be incorporated in the Deluge browser at a future date

m-m-adams, 2024-10-06:
This seems to have largely worked out through people naming synths the new way over time

closed 2024-10-06 by m-m-adams