SynthstromAudible/DelugeFirmware #670: MIDI CC parameter jumps to maximum value when turning gold knob

sdesign75, 2023-11-01:
### Please describe the problem:

Cannot use gold encoders to control MIDI CC’s. After assigning a midi CC to a gold knob, when you turn the knob it jumps to the maximum value and cannot be changed.

### What is the expected behavior?

Gold knob changes the midi cc parameter

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

community

### What is the firmware name:

1.0.0-BETA-A258921

sdesign75, 2023-11-02:
Can anyone confirm this bug?

m-m-adams, 2023-11-03:
fixed by #678

closed 2023-11-03 by m-m-adams

sdesign75, 2023-11-03:
Awesome!