SynthstromAudible/DelugeFirmware #673: Sync-Scaling not functional

sdesign75, 2023-11-01:
### Please describe the problem:

Button does nothing when pressed

### What is the expected behavior?

It should enable sync scaling

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

community

m-m-adams, 2023-11-02:
Go to community features and change the sync scaling option. More info can be found here: https://github.com/SynthstromAudible/DelugeFirmware/blob/community/docs/community_features.md#434---fill-mode

closed 2023-11-02 by m-m-adams

sdesign75, 2023-11-02:
Thanks, I didn’t realise it had been overridden :-)