SynthstromAudible/DelugeFirmware #685: deluge community build A77E2DE hard crash in settings menu

duncanrmi, 2023-11-05:
### Please describe the problem:

in the settings menu, attempting to select either 'sticky shift' or 'light shift' crashes the unit.

machine is OLED, build is deluge-v1_0_0-beta+2023_11_05-a77e2de

### What is the expected behavior?

sticky shift/illuminated shift button

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

community

### What is the firmware name:

deluge-v1_0_0-beta+2023_11_05-a77e2de

### If possible provide the steps to reproduce the issue and upload additional media:

turned on, entered settings menu/community build, scrolled down to either of the shift key options, attempted to alter settings.

after a few seconds, a pattern appears in the grid [pic].
[image]

alifeinbinary, 2023-11-07:
Noticing this issue as well with firmware 1.0.0-beta-5A9C21D. The Deluge also locks up if you try to scroll past LIGHT SHIFT. @m-m-adams suspects it might be related to the type of encoder, so I'll add that my unit is a retrofitted 7SEG era model from 2018. When did you buy yours @duncanrmi ?

m-m-adams, 2023-11-07:
Fine until https://github.com/SynthstromAudible/DelugeFirmware/commit/390630a47bb2de655a1ac37a3137d5cbbb145545

duncanrmi, 2023-11-07:
april 2018, is my original unit; the unit affected is my second one though, with a slightly later s/n. (bought used)
still 2018 I’d say.
both OLED.

duncanrmi, 2023-11-07:
180421 & 180505; this latter has the issue. both (presumably) 2018 machines- is that the date in the s/n?

m-m-adams, 2023-11-07:
Fixed by #694

closed 2023-11-07 by m-m-adams