SynthstromAudible/DelugeFirmware #692: Scrolling thru the songs list is not working properly

Balistix55, 2023-11-06:
### Please describe the problem:

If I want to load a song from a list of songs on sd card (for example 25 songs) and scroll thru the list from the top, after 5 or 6 names there is a blank row. If I want to scroll below this blank row it takes me back 3 songs before and it is impossible pass down and scroll down to the bottom. But if I scroll from the bottom to top, it is possible to scroll thru all songs on sd card.

This issue only appeares in the main song folder. If I choose subfolder and scroll the song thru subfolder, then scrolling works well. 

### What is the expected behavior?

Scrolling in the list of songs in the main folder should work exactly the same as in official release 4.1.3 

### Is there a relevant Pull request?

no

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

nightly, community

### What is the firmware name:

deluge-v1_0_0-beta+2023_11_05-5a9c21d

### If possible provide the steps to reproduce the issue and upload additional media:

https://github.com/SynthstromAudible/DelugeFirmware/assets/137617365/09792004-3dd8-4639-96ed-efb2d66b7421

duncanrmi, 2023-11-06:
can you still access them all on your computer via a card-reader? there might be a corrupted file in your list that the machine is struggling to parse.
you can inspect the xml files in a text editor. my bet is you'll find one that won't open- bad card? interrupted save? a song from a different fw version? 
I'm running this release here, no issues with song load yet- deluge-v1_0_0-beta+2023_11_05-a77e2de

Balistix55, 2023-11-06:
I checked all files in SONGS folder. One file on sd card was corupted, I deleted it and also other subfolders with songs, that were made in different firmware version. But issue is still there. 

If I scroll backward, I can scroll all files in folder from bottom to top.

In official 4.1.3 scrolling thru files works properly.

duncanrmi, 2023-11-06:
it’s beginning to sound like the card may be the issue, then, if you found a corrupted file. 
could you see what was wrong with it? broken xml?

but at this point, I’d be inclined to back up the remaining files & reformat the card, or maybe try a new one.

I’ll fire mine up now on the version I mentioned (I have a second OLED machine running factory, with the cards kept identical by a copy-routine on my computer) & see if it’s still ok.

Balistix55, 2023-11-06:
It was xml file, that was not able to open. I could check another sd card, also reformat actual one. I will post conclusion then.

Balistix55, 2023-11-07:
@duncanrmi it looks, you were right with corrupted files on sd card. I did not find them, but deleted folders with collected songs and scrolling is without problem. I also tried another sd card and is ok too. So do I need to delete this issue? Maybe it can help you in the future with similar questions.

closed 2023-11-07 by PaulFreund

duncanrmi, 2023-11-07:
admin will close it- it moves to a different section & will remain searchable to help others!