SynthstromAudible/DelugeFirmware #693: Master compressor basically just distorting

tobiasphilippen, 2023-11-06:
### Please describe the problem:

I find it almost impossible to make the master compressor sound good. It starts to distort very quickly, and changing the attack and release settings does not change that significantly. Also, even without any LPF engaged, volume jumps can be very abrupt. I find the compressor basically unusable as it is. Settings also used to be not saved properly, so that when opening a song, compressor setting was 50, and the sound was completely distorted. This seems to have been fixed in the current firmware version.

Also, settings from 0 to 50 for ratio do not make sense.

The compressor implemented by AlterAlter seemed to work fine and was well done from an engineering point of view. Maybe this one could be incorporated again?

### What is the expected behavior?

Compressor not distorting at all if set to less than 50 with a reasonably long release time.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

community

### What is the firmware name:

1.0.0-BETA.5A9C21D

m-m-adams, 2023-11-06:
It was not well done from an engineering point of view, it used approximately 15 voices worth of CPU and did not integrate to any of the deluge systems. Try turning down the song level and compressor threshold, both will lead to less distortion.

The actual ratio is from 2 to 20 it's just displayed as 0-50 like all other deluge parameters 

What is the issue with the LPF?

tobiasphilippen, 2023-11-06:
What I meant by engineering is *audio* engineering. Of course, I would not know about how it was done in terms of programming. And I have never used it, just watched the video. So, no, not saved anything with settings from the old compressor.
As to ratio: Well, pitch for example is displayed in semitones and cents – so that the numbers have a musically relevant meaning ... might make sense to have the usual ratios and milliseconds for attack and release, too.
Audio levels are already quite low, but I can try that. Why would lowering the threshold lead to less distortion? When I turn up the upper dial, will this lower the threshold and increase the make up gain at the same time?
Concerning LPF: I just read that there seemed to have been a distortion issue if compressor and LPF was engaged at the same time ...
Thank you.

m-m-adams, 2023-11-06:
Yes, there is auto makeup gain tied to the threshold and ratio. Due to popular demand to be able to clip, if the song level is above 30 or so it can clip the output. I'm open to tweaking that

The lpf and compressor interaction is fixed, it was a bug in the previous compressor and one of the reasons for the change

m-m-adams, 2023-11-06:
The song level is also post compressor, lowering it should get rid of any clipping to

tobiasphilippen, 2023-11-07:
By song level, I suppose you mean adjusting the "level" parameter when in song mode, right? Yes, I noticed it is post, because it does not have any effect on the compression. I don't know whether that makes sense, as the master compressor might be understood more like a limiter, which is usually post the master fader. Anyway, the only thing then to get a less hot signal going into the compressor would be to turn down all tracks, which is a bit impractical. And the clipping then would be from the D-A converters?
Well, maybe it is just me, but I am struggling to get a decent sound out of the compressor. When the upper dial is above 12 or so, distortion already starts to set in, and it does not have a pleasant analog character to it, more like digital distortion. The attack control is not really able to shape the transients in the way that is typical for a master buss compressor, and when the setting is 50, I am getting extremely jumpy results. Also, the release should make it possible to make the compressor move with the groove of the track, but changes are not really audible except for the extreme settings which are distorting in the low values and choking the track nearing 50.

I do not want to sound ungrateful. I am indeed very grateful for all the work you have done, and the community firmware is already a huge step forward and overall a joy to use. Is it possible that the compressor is just a little bit too much focussed on using as little resources as possible so that it falls short in terms of resolution and audio quality? Are other users satisfied with the results they are getting? As a mixing engineer, I am used to both plugin and hardware compressors on the 2 buss, so I am confused that this master compressor is so difficult to dial in ...

tobiasphilippen, 2023-11-07:
... yeah, thinking about it: song level post compressor does make sense, though.

m-m-adams, 2023-11-07:
Yeah the clipping is at the 32 to 24 bit conversion prior to the output. I'd originally set it so the compressor would never cause clipping but there were lots of people who wanted the clipping added back in. As is moving the volume post comp so it would work as an overall volume again and allow for reducing the gain to not click seemed like the best compromise

I'll take another look at the attack/release settings, the render window size can be as small as a single sample after the other optimizations and bug fixes last week and that might have a buggy interaction with the compressor

m-m-adams, 2023-11-07:
If you want to join our discord I'm totally happy to discuss tweaks before release - https://discord.com/channels/608916579421257728/1107026299945496577

tobiasphilippen, 2023-11-07:
Hey, thanks. I am happy to join on discord, but it does not work with this link ... it says, I do not have access.

m-m-adams, 2023-11-07:
I might have made the link incorrectly, try this one https://discord.gg/24mpbKjhJ6

m-m-adams, 2023-11-09:
Completed in #695

closed 2023-11-09 by m-m-adams