SynthstromAudible/DelugeFirmware #706: holding  shift, select synth mode, while holding shift select synth at other end of list hags every time

j0vius, 2023-11-09:
### Please describe the problem:

I've tried this new song , new synth , shift synth mode , while holding shift scroll through option , hangs , normally on ring but depends what is selected when i star to scroll .
i know you should let go of shift but if you've got it held it hangs it 

### What is the expected behavior?

have to power cycle to get functionality again 

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

nightly

### What is the firmware name:

deluge-v1_0_0-nightly+2023_11_08-e614a08

PaulFreund, 2023-11-09:
Can you please check with todays beta build? A PR was merged to address menu scrolling :)

j0vius, 2023-11-09:
> Can you please check with todays beta build? A PR was merged to address menu scrolling :)

yes still does it

j0vius, 2023-11-10:
ok  so with the beta it doesn't hang , holding shift you can only select the top or bottom menu item but apart from that all works as should

closed 2023-11-10 by PaulFreund

m-m-adams, 2023-11-10:
Yeah, shift increases the scrolling speed to 5 at a time. This is intended behaviour