SynthstromAudible/DelugeFirmware #715: Unable to assign to kit row a synth that was just created

laikrodiz, 2023-11-12:
### Please describe the problem:

When I create a new synth, save it then assign to a kit row, the sound is incorrect.

https://youtu.be/kzVpA7LVvV0

### What is the expected behavior?

Synth sound assigned to kit pad reproduced correctly

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

community

### What is the firmware name:

deluge-v1_0_0-beta+2023_11_10-e671767

### If possible provide the steps to reproduce the issue and upload additional media:

1. Create new synth
2. Tweak its settings
3. Save it
4. Create new kit
5. Create a synth pad in kit
6. Select the synth patch just saved and play it

The synth in kit sounds completely different from the one I just created.
The bug persists no matter if you save it as numerals or give it a unique name.

m-m-adams, 2023-11-12:
I'm probably just going to try giving a better error message for this release - as far as I can tell it's a timing issue with when synths actually get written to the sd card

m-m-adams, 2023-11-13:
note for later - it's this line causing issues https://github.com/SynthstromAudible/DelugeFirmware/blob/8bff6c0f86ed92429e99353d6f36ca17a3cc2ae5/src/deluge/gui/ui/browser/browser.cpp#L545

PaulFreund, 2023-11-17:
Is linked to #718 in both cases we end up with a file entry in the list without a file pointer

closed 2023-11-26 by m-m-adams