SynthstromAudible/DelugeFirmware #727: Reorganize Community Features by functional areas, create release notes

m-m-adams, 2023-11-18:
### Please describe the problem:

The community features list is chronological - this works for people who have been actively following development, but will be confusing for people who first look at the firmware post release.

We should reorganize into functional areas:

- Sound (filters, wavefolding, compressor)
- Sequencer (Automation, quantize+humanize, row probability, fill mode)
- QoL stuff (sticky shift, loading synths into kits)
- Midi (per device clock, more virtual usb ports)
- Bugfixes

### What is the expected behavior?

Someone new to the community release has an easy way to find things they might like

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

nightly

m-m-adams, 2023-11-18:
@ok-reza I assigned you for now but feel free to change that

ok-reza, 2023-11-18:
i could help out here, however it has been helpful for me to know that the last entries in each category of communityfeatures.md are the newest. is there a clean solution to having dates incorporated ? either including dates at the end or creating a separate list of the features sorted by date w no descriptions?

m-m-adams, 2023-11-18:
Yeah keeping the current one and renaming it to changelog or something is a good idea

PaulTobia, 2023-12-06:
I would recommend categories aligned with the official guidebook so documentation has some consistency between community and official?
- BASICOPERATION
- SEQUENCING
- SYNTHESIZERS
- KITS
- MODULATION
- SONGVIEW
- ARRANGERVIEW
- AUDIO
- LOOPING
- EFFECTS
- MID
- CV
- SYSTEM&GENERAL

closed 2024-01-08 by m-m-adams