SynthstromAudible/DelugeFirmware #731: Freezing randomly changing tempo

rpc9943, 2023-11-20:
### Please describe the problem:

Freezes when changing tempo

### What is the expected behavior?

Not freezing

### Is there a relevant Pull request?

No idea what that means

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Built from community

### What is the firmware name:

1.0.0-beta-dba3e3b

### If possible provide the steps to reproduce the issue and upload additional media:

Turn on set to any synth preset, turn tempo up and down. It randomly freezes, tried default resolution 192 and 6144 same issue. Happened about 6 times+

PaulFreund, 2023-11-21:
Fixed by #735

closed 2023-11-21 by PaulFreund

rpc9943, 2023-11-22:
Sorry in the beta im still getting freezing immediately after turning on and changing tempo, once at 2k bpm another at 4k. Version 1.0.0-beta-D5331ED