SynthstromAudible/DelugeFirmware #741: Solo function not present in grid mode

rpc9943, 2023-11-22:
### Please describe the problem:

Solo is not working on grid mode. Works on rows.

### What is the expected behavior?

For grid to be able to use solo function as it works in rows

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Built from community

### What is the firmware name:

1.0.0 betaD5331ED

### If possible provide the steps to reproduce the issue and upload additional media:

Hold leftright knob down press the clip on grid mode. It does not solo. It works this way in rows view where you press the mute launch pad.

m-m-adams, 2023-11-22:
Honestly didn't know that shortcut existed

PaulFreund, 2023-11-22:
This is a bug if select in green mode setting is active, need to fix it

commit 4c959c66 (2023-11-23): Fix #741 and #743 - Horizontal encoder button in Grid mode (#748)
closed 2023-11-23 by PaulFreund