SynthstromAudible/DelugeFirmware #746: Auditioning note with Arp ON won't reflect dotted/triplet sync rates.

ok-reza, 2023-11-22:
### Please describe the problem:

Set ARP SYNC to 16th.
Hold Audition pad, or a note in Keyboard view.
Switch ARP SYNC to 16th-triplets.
Hold Audition pad, or a note in Keyboard view.
Note: The same rate is played back despite different rates.

Sequence a note for the full length of the bar.
Press Play
The proper sync rate, 16th-triplets, is played back.

### What is the expected behavior?

The rate ARP SYNC is set to should be reflected on auditioned pads. Currently it will treat any triplet or dotted sync rate as a normal straight division.

### Is there a relevant Pull request?

#32

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Built from release/1.0

### What is the firmware name:

beta-d5331ed

PaulFreund, 2023-11-22:
Needs to be analyzed and categorized before agreeing on a milestone

sapphire-arches, 2024-02-04:
@ok-reza can you still reproduce this? I wasn't able to, using the "021 80s TV LEAD" factory synth, or an init synth.

ok-reza, 2024-02-23:
@sapphire-arches yep! same issue and was able to reproduce on any synth preset including 021 80's TV LEAD.

the simplest check is to create a drone note for 1 bar (let's say a C note), have arp set to 16th-triplet rate. play the sequence and make a mental note of the rate. now stop playback and press and hold a C note from the audition pads. the audition pad will resemble a 16th rate arp, not a 16th-triplet rate.

closed 2024-03-21 by m-m-adams