SynthstromAudible/DelugeFirmware #747: Ring Mod Synth with Noise will produce harsh noise with Unison Stereo Spread and Unison

ok-reza, 2023-11-22:
### Please describe the problem:

*Warning, extremely loud noise is produced, highly suggest lowering Output Level to like 5%*

Super loud noise is produced when a ring mod synth has noise > 0, unision stereo spread > 0, and unison number > 1.

### What is the expected behavior?

No jump scares.

Unsure if this is the nature of stereo noise and ring mod in conjunction or just a bug.
That being said, you can create some wild sounds messing with this, HPF/RES, wavefolding, morphing, adjusting detune/number/stereo spread.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Built from release/1.0

### What is the firmware name:

beta-d5331ed

### If possible provide the steps to reproduce the issue and upload additional media:

Create New Synth
Set SYNTH MODE to RINGMOD
Set UNISON NUMBER to greater than 1; lets do 3
Set UNISON STEREO SPREAD to greater than 1; lets do 2
Current sound is fine, audition note.
Raise NOISE to greater than 0; 1 is fine.
*LOUD WARNING* Lower Output Volume to 5% before auditioning.
Note the craziness.

Adjusting Unison Number to 2 creates a more tamed audio noise/hiss.

PaulFreund, 2023-11-22:
Needs to be analyzed and categorized before agreeing on a milestone

PaulFreund, 2023-11-26:
I don't think we will fix it before release but we should put it in a "known issues" section in the release notes that this should not be combined

PR #1168 (2024-02-04): voice: fix noise+stereo spread going out of control
Fixes #747
commit 64e92131 (2024-02-04): voice: fix noise+stereo spread going out of control (#1168)

Fixes #747
closed 2024-02-04 by m-m-adams