SynthstromAudible/DelugeFirmware #751: Solo function in grid mode needs visual indicator

rpc9943, 2023-11-23:
### Please describe the problem:

Compare song "rows" view solo vs grid, we can see the blue as an indicator, whereas solo on grid we see the active pads lit and the ones that will play when we disable solo, but we probably should be able to see the soloed pads as blue. It can become confusing if we do not have this visual cue, maybe even a slow blinking blue might work on top of the current setup.

### What is the expected behavior?

Consistency between solo in rows vs grid, blue pads for solo 

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Built from release/1.0

### What is the firmware name:

1.0.0 beta 4c959c6

### If possible provide the steps to reproduce the issue and upload additional media:

Solo grid pad using push LR knob and pad to solo

PaulFreund, 2023-11-23:
Just showing as blue does not work visually, I tested it and it is a lot more confusing, I'll leave this open to see if somebody has a good suggestion we can try. Pulsing was suggested but it collides with arming flash and I think even making it slower will be a little strange for solo

Quadristi-Riczi, 2023-12-07:
> Just showing as blue does not work visually, I tested it and it is a lot more confusing, I'll leave this open to see if somebody has a good suggestion we can try. Pulsing was suggested but it collides with arming flash and I think even making it slower will be a little strange for solo

Do you think it would help to dim the other pads as well as show blue?

riczikovacs, 2023-12-07:
> Just showing as blue does not work visually, I tested it and it is a lot more confusing, I'll leave this open to see if somebody has a good suggestion we can try. Pulsing was suggested but it collides with arming flash and I think even making it slower will be a little strange for solo

Do you think it would help to dim all the other pads as well as show blue?

sichtbeton, 2024-03-22:
There is a dimming going on when soloing in the latest nightly. Could be closed?

closed 2024-12-28 by m-m-adams