SynthstromAudible/DelugeFirmware #761: Gold knob value pop up is on the wrong side on 7seg

m-m-adams, 2023-11-26:
### Please describe the problem:

Pop up should be in right two digits, instead it’s appearing in the left two. Currently this causes flickering as menus and note popups render in the right digits

### What is the expected behavior?

Right digits to match menus and not conflict with note pop up.

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

Built from release/1.0

PaulFreund, 2023-11-26:
Is this something for future release of do you want to fix during RC?

m-m-adams, 2023-11-26:
It'll depend how easy the fix is - I haven't looked at it at all yet

Maybe @seangoodvibes knows?

closed 2023-11-27 by PaulFreund