SynthstromAudible/DelugeFirmware #785: Drum keyboard mode responsivity issue

Maj7sus2, 2023-12-05:
### Please describe the problem:

Tempo set at 120 BPM: while repeated playing with two fingers is fairly responsive, trying to match a continous 16ths chain over 4 different velocities of the same sample seems to trigger some issues, with one of the inputs simply being skipped.

Not sure if it's a bug or I just hit the limit ( triplets at 120 bpm are absolutely no problem).

### What is the expected behavior?

Best case scenario: lagless response to the inputs on a new blank project.

realistic scenario: some minor lag, especially in already crowded projects.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

community

### What is the firmware name:

amadeus

### If possible provide the steps to reproduce the issue and upload additional media:

New Song
Drumkit (I use 000 Tr-808) in keyboard mode
pick any sample and try to play 16ths continously over the 4 highest velocities (I  tried both with two fingers per hand, 3+1 and one handed)
If you wanna make sure, turn on metronome (120 BPM) and try to play 16ths. if you try to play triplets on 120 bpm, the response is flawless and would make the issue more noticeable.

PaulFreund, 2023-12-05:
I tested this with a BPM of 178 and I can play a four to the floor (16th notes) with the four highest velocities without problem. To test my accuracy I disabled record quantization and I was 1/128th note off for three of the beats but I can't do any better with the sidebar in clip view. Maybe you could provide a video?

Maj7sus2, 2023-12-06:
> this is because if you hold down a note while trying to press another note of a different velocity, it won't retrigger since a note of the same sample is still being held down.
> 
> currently a limitation of the drum keyboard view i believe.

Yeah I think this is the issue! I guess it's not really a bug then, but it does kind of limit the performance imho. I am used to finger drum on the launchpad and on Deluge i couldn't really make much happen lol.

Another cool feature would be to be able to set the size of the squares  (with less velocities per sample), but this is just me making wishes now XD
Thank you for your help!

ok-reza, 2023-12-06:
i imagine it will be addressed at some point, iirc it is a little complicated to implement 

> Another cool feature would be to be able to set the size of the squares (with less velocities per sample), but this is just me making wishes now XD Thank you for your help!

that's actually currently possible, hold SHIFT + turn <> encoder. can shift from 1x1 to 8x8 squares.

Maj7sus2, 2023-12-06:
Well, if I did not learn something today. Thank you again!

m-m-adams, 2023-12-11:
This is a deluge limitation in general, you can never have more than 1 sustained note in a row 

Fixing it isn't feasible without a ground up rewrite of how instruments work sadly

closed 2023-12-11 by m-m-adams