SynthstromAudible/DelugeFirmware #786: drum kit - load random sample causes crashes / freezes the deluge

DelayDude, 2023-12-05:
### Please describe the problem:

I´ve loaded the first factory kit and randomized the kick and snare. That works great but as soon as I´ve randomized the hat, the deluge freezes. I have to push the select button, to recover it. 

### What is the expected behavior?

no crashes when randomizing the third sample sound  :-)

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

community

### What is the firmware name:

1.0.0

DelayDude, 2023-12-06:
Edit: only happend once.

m-m-adams, 2023-12-06:
Do you remember what the freeze message was?

DelayDude, 2023-12-07:
I don´t remember the error code number . But the display said something like: "press select to attempt resume. Then save to a new file."

scnroy, 2023-12-12:
Not sure if this is the same issue, but I can reliably get the Deluge to crash (random LEDs light up and bzzzzzzzzzz sound—no error message on OLED) if i do randomize drum on a kit using synth drum sounds (in my case the Drummage pack). 

Happy to open another issue if it'd be easier to track that way.

m-m-adams, 2023-12-12:
I think that sounds different - could you open another one?

closed 2023-12-20 by m-m-adams