SynthstromAudible/DelugeFirmware #787: Scrolling through presets gets laggy (CPU?) when 8 initalized synth clips are created

ok-reza, 2023-12-05:
### Please describe the problem:

creating new unique synth clips, then initializing each of them will create a CPU slow down. this is visible when choosing one synth clip to scroll presets with via SELECT and noticing lag. can occur with as little as 4 init clips.

notes:
on 7SEG: initialized presets all have the name '0' and do not increment (separate but related issue?)
on OLED: initialized presets all have the name 'SYNTH#', increment, but i believe is different from official where it should be 'SYNTH###' (not confirmed).

### What is the expected behavior?

confirmed with v4.1.4-RC, no slow down when scrolling through presets while in Song view should occur with 8 init synth clips present.

### What hardware did you reproduce it with?

OLED, 7SEG

### What firmware did you reproduce it with?

Built from release/1.0

### What is the firmware name:

c1.0.0

caseygipson, 2023-12-06:
I'm just now noticing laggy synth scrolling on a new project with 5 newly initialized wavetable synths and 1 newly initialized kit instrument. The problem does not go away after a reboot and settings > firmeware > Y, available memory looks the same on initial load as a brand new project.

m-m-adams, 2023-12-06:
I believe this comes from the fix for #715 and #718 requiring extra sd reads 

Hopefully I can come up with a better fix going forward

dzaxy, 2023-12-18:
synth kit preset wavetable scroling is cryptonite for my oled deluge too

SimonSaysBoo, 2024-01-07:
Yes, I've got laggy synth preset scrolling on a fresh song with nothing entered

sichtbeton, 2024-03-22:
Is this still the case? At least here not anymore in latest nightly. Could be closed?

seangoodvibes, 2024-03-29:
> Is this still the case? At least here not anymore in latest nightly. Could be closed?

Yea it is still the case.

If you create a synth clip with an arp and set a rhythm param value it's pretty noticeable.

m-m-adams, 2024-04-04:
That's definitely a different problem. This issue arises when you have multiple presets in your song that don't exist on the sd card, and comes from unnecessarily checking to see if they exist on the card

PR #1783 (2024-04-27): cache whether the instrument is on card to avoid re checking
Fix #787 

Save whether an instrument exists on card or not, and only grab the file pointer if it exists but we don't know it
PR #1785 (2024-04-27): cache whether the instrument is on card to avoid re checking
Fix #787 

Save whether an instrument exists on card or not, and only grab the file pointer if it exists but we don't know it
closed 2024-04-27 by m-m-adams