SynthstromAudible/DelugeFirmware #792: Song loading issue

DelayDude, 2023-12-06:
### Please describe the problem:

I´ve saved my first song with the Amadeus firmware. When I´ve loaded another song and want to load this first song again, the file is not found (when I enter the name with the keyboard). I can find the song in the list. 
And the compressor switches back to "one" even if I´ve saved the "full" compressor. 

### What is the expected behavior?

Enter the song name and load it from the list. 

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

community

### What is the firmware name:

Amadeus

### If possible provide the steps to reproduce the issue and upload additional media:

1. Save song in "Amadeus"
2. load another song
3. click load and enter the songname of the first song
4. file not found

Icoustic, 2023-12-07:
So to be clear you can find the song when scrolling manually though?

DelayDude, 2023-12-08:
Thanks for your reply! What I meant is, that (even in "one" mode) the compressor sometimes did not work.

m-m-adams, 2023-12-08:
I believe it is your compressor settings, one mode doesn't change anything and still uses the settings you chose in full. If you save a song where you feel the compressor isn't working I can take a look

closed 2023-12-11 by m-m-adams