SynthstromAudible/DelugeFirmware #798: Grid view does not display all clips

gobcmmelo, 2023-12-07:
### Please describe the problem:

Switching to grid view does not display all clips in from rows view. In this particular case this has occurred with 2 clips using the same instrument, the first row appears but additional instruments below the first row of the column do not appear until the row is pressed.

### What is the expected behavior?

Grid view displays all clips and multiple clips that use the same instrument are displayed in a single column.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Built from release/1.0

### What is the firmware name:

C1.0.0

### If possible provide the steps to reproduce the issue and upload additional media:

Created a song with a multiple clips using the same instrument that are not linked. Switch to grid view.

ok-reza, 2023-12-08:
I believe this is a result of the two same-instrument clips sharing the same section color. Although Rows View allows for that, it's not exactly clear in Grid View how that would be handled in a way that is clear/makes sense. This is a currently known limitation unless you are describing something different.

gobcmmelo, 2023-12-08:
> I believe this is a result of the two same-instrument clips sharing the same section color. Although Rows View allows for that, it's not exactly clear in Grid View how that would be handled in a way [...]

That appears to be it. Thanks, I guess this would be categorized as an enhancement then. Thanks!

m-m-adams, 2023-12-11:
Closing the issue, but feel free to open a discussion if you have ideas for how to display the different variants in grid view!

closed 2023-12-11 by m-m-adams

gobcmmelo, 2023-12-11:
Totally makes sense, thanks!

> Closing the issue, but feel free to open a discussion if you have ideas for how to display the different variants in grid view!